GeoNet/help #80: Post-processing of waveform data

mayjinteoh, 2021-01-29:
Hi team,

I'm trying access ground motion(waveform) data from dataobspy.core.trace.Trace class and convert it to acceleration response spectrum. 
1. I was wondering if the ground motion(waveform) data I'm getting from the obspy class includes correction for the dynamic instrument response, high- and low-pass filtering? 
2. If not, what frequencies would be appropriate to produce acceleration response spectrum?
3. The magnitudes of waveforms I'm getting do not seem to be consistent from event to event, would you able to advise what unit is the waveform recorded at? I'm hoping to convert them to m/s^2.

Thank you very much.

May

salichon, 2021-07-04:
FYI @JonoHanson 
(perhaps this type of ticket  also could be co- handled through the "info" parts (with sara)  
(we are slightly lagging in time :) )

staylorofford, 2021-07-04:
@mayjinteoh apologies for the delay in responding.

obspy can be pretty opaque, I have had similar questions myself and had to look over the source code to find answers. In all honestly, I only found them recently!

1. If you're querying using the `get_waveforms` method of the FDSN client (imported as something like `from obspy.core.fdsn import Client`), then you will be getting unmodified data from GeoNet's archive. The only filtering is what is done naturally by the sensor by its response. You can include the response information by using a function call like:
```
client = Client('GEONET')
stream = client.get_waveforms(network=network, station=station, channel=channel, location=location, starttime=starttime, endtime=endtime, attach_response=True)
```
which will give you the data and the combined sensor + digitiser response. You can then remove the response with:
```
stream.remove_response()
```

2. You will want to examine the response to know which are suitable. From a data perspective, the frequencies over which the data is clearly recorded (let's say response over 0.1) would be safe, with those with lower response more likely to introduce noise.

3. The waveform will be in sensor counts * gain. To convert to m/s (seismogram) or m/s^2 (accelerometer) you can do this approximately by removing the gain (sensitivity in response information), or you can do this accurately by removing the response. It should be as easy as how I present it in 1.

Let me know how you get on, this is a tricky problem!

Thanks,

Sam Taylor-Offord

closed 2021-10-04 by salichon