GeoNet/help #81: GNS Contact

Ernesto03-codes, 2021-03-17:
Guys

Who should I contact regarding some queries I have regarding https://github.com/GeoNet/data-tutorials? I've run the code using Jupyter notebook to extract the PGAs of certain stations depending on the EQ events. Just wanted to confirm if what I did is correct or not.

If you could please contact me.

Cheers
EQC Engineer

elidana, 2021-03-18:
Dear Ernesto,

thanks for reaching out. Are you happy to continue the conversation in this ticket? It might potentially interest other GeoNet data users.

Do you have a specific tutorial in mind? We are currently reviewing those, as some of them might need updates and such.

If you rather prefer having an email conversation, can you please send an email to `info@geonet.org.nz`?

Thanks,

Elisabetta

elidana, 2021-03-29:
Hi @Ernesto03-codes ,

thanks for contacting us directly. I am closing this issue as we are continuing the conversation on other channels.

closed 2021-03-29 by elidana