GeoNet/help #82: Is FOZ connected to the seedlink service?

calum-chamberlain, 2021-07-08:
For a while now I have not been getting any data from FOZ via seedlink - I don't run seedlink queries often enough to know if this is a long standing issue. 

For example the following code returns traces:
```python
from obspy.clients.seedlink.easyseedlink import create_client

def handle_data(trace):
    print('Received the following trace:')
    print(trace)

client = create_client('link.geonet.org.nz', on_data=handle_data)

client.select_select_stream("NZ", "RPZ", "HH?")
client.run()
```

in contrast, selecting channels from FOZ does not return any data:
```
client = create_client('link.geonet.org.nz', on_data=handle_data)

client.select_select_stream("NZ", "FOZ", "HH?")
client.run()
```

fingers crossed that this a transient issue, but I have haven't seen FOZ data for the last couple of weeks.

elidana, 2021-07-08:
Hi @calum-chamberlain ,

thanks for your question!

As far as I know, FOZ site has been offline since the 16 of June. If you wish, we can endeavor to let you know when the site will be back.

@salichon might be able to help answering the `seedlink` part of the question better than me.

calum-chamberlain, 2021-07-08:
Ah, that would do it! 
I think I'm happy with the seedlink part if FOZ is offline.

closed 2021-07-08 by calum-chamberlain

elidana, 2021-07-08:
brilliant thanks! 
I think we are beating a record here with our tickets, opened and closed in 20 minutes. Something to celebrate 😆