GeoNet/help #83: RAKW GNSS time series- vertical component degradation

elidana, 2021-09-23:
The GNSS site RAKW is showing a degradation of time series data after 22 April 2021 (Sigrun picked this up and notified us).

> Do you know what happened to RAKW around the 22 April this year?
> Data look at bit messy and there is a 5 cm subsidence at the site. 
> Looks like something happened to the antenna. I suspect the antenna is broken and would need the quality plot data to check that

@megankortink confirmed that there are no equipment change that can explain that or issues with the site recorded in the ticketing system, and that the antenna at the site is +10years old. 

> It could be the antenna it looks quite old ….
> `Trimble Navigation Ltd.,TRM57971.00,1441031099,RAKW,0.002,0,0,0,2011-02-16T04:05:01Z,9999-01-01T00:00:00Z`

Land ownership changed around the same period, so we will investigate if the vertical offset and poor quality of data after April 22 might have been caused by a change in the antenna surroundings.
Another possible options is that the antenna is degrading, as it was installed in 2011 

A link to the vertical time series plot:
https://fits.geonet.org.nz/plot?type=scatter&sites=RAKW&typeID=u&start=2020-01-01T00:00:00Z&days=700

megankortink, 2021-09-23:
Some basic plots for cycleslip and mp1/mp2 from the 1st April 2021 until 1st June 2021, initial testing, can test with a longer duration
[image]
[image]

Mp1 (0.21 --> 0.25) and Mp2 (0.48 --> 0.54) looks like values are slowly increasing, cycle slip fairly consistant and fluctuating

megankortink, 2021-09-27:
Data for Mp1, Mp2, Cycleslip from gloria

[image]
[image]

elidana, 2021-10-21:
We have removed the antenna and installed a new one today. 
No water damage was found in either antenna cable ends, or the TRM57971.00 that was installed at the site.
We have also performed a tie in survey to have a better control on the offset introduced by the antenna swap.

Will revisit time series data in 2 weeks time to (hopefully) verify an improvement in data quality

closed 2023-10-18 by elidana