GeoNet/help #84: MQZ issue with broadband vs. accelerometer comparison

elidana, 2021-09-24:
Jesse Hutchinson (University of Canterbury) has noted an issue with MQZ seismic data when comparing data from the broadband and strong motion sensors installed at the station. 

His method calculates the difference of the natural log values of PGAStrong from PGABroadband instruments, for all M>5 events recorded by the station.

A plot he provided is showing a consistent discrepancy for all events recorded between ~2014 and ~2020.

[image]

Further investigation to understand the cause are needed

salichon, 2021-09-27:
- i  encourage to have a  look at the raw data  (say using daily streams) as it should be done prior to any processing of some sort  :) - especially  when it is made with big batch (though tedious i know)

salichon, 2021-09-29:
--Summary of the PSA process above -- (perhaps to polish) from what i understood

_Input_:
-  M5 event waveforms  (time window length?)
- Waveforms from high quality station (national sites)
- Use in priority  strong motion HN (200HZ)  (then strong BN,  and then weak  HH 100Hz)

_Process_: 
- Remove isntruments response
- Compute  _**P**seudo-absolute response **S**pectral **A**ccelerationsresponse spectra_ (PSA), 
From:
  1 - Strong motion 
  2 - Weak motion  
    --> PSA methods fro each data type  to be desccribe briefly  
    
_Tools and Sources_: 
- Used Python package to [perform this computation

salichon, 2021-09-29:
MQZ strong motion periods  (delta/fdsnwebservice:  https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=MQZ&level=channel&format=text&channel=?NZ)

- NZ|MQZ|20| HNZ/BNZ |Accelerometer|427336.117780|1.000000|m/s**2|50.000000| **2009-08-06T00:00:01|2016-02-22T20:00:00**
- NZ|MQZ|20|HNZ/BNZ|Accelerometer|427336.117780|1.000000|m/s**2|50.000000| **2016-02-24T01:02:00|2019-06-17T05:25:00**
- NZ|MQZ|20|HNZ/BNZ|Accelerometer|427818.982500|1.000000|m/s**2|200.000000| **2019-06-17T23:20:00| NOW**

SquirrelKnight, 2021-10-01:
The procedure is as follows:
All stations for which an event arrival was detected and within a certain distance defined by a magnitude-distance scale are used to download waveforms. The time window extends from 10 seconds prior to the P arrival to the S arrival + the ds595 value * 1.2. The ds595 value has been modified from the original equations by Afshari and Stewart (2016) to include an additional term to capture more of the waveform coda for larger magnitude events.

After the raw mseed files are downloaded, they are then evaluated with a ground-motion classifier to determine whether the event meets certain quality criteria. High quality events are then pre-processed for intensity measures. In pre-processing, I first demean the data and pad the data for five seconds on either end. I then remove instrument sensitivity and differentiate any velocity data into acceleration. The data are then rotated to ZNE components and trimmed back to their original window length. Finally, the data are low-pass filtered at a corner frequency of 0.05 Hz.

The pre-processed data is used to compute intensity measures (PGA, PGV, pSAs, etc.) for horizontal, vertical, rotd100, and rotd50 components.

The list of M5 events used for MQZ are as follows (note that this analysis has only been performed between HN and HH channels):
'2016p858069', '2016p858147', '2016p858156', '2016p858160','2016p858279', '2016p858848', '2016p858895', '2016p858951','2016p859051', '2016p859105', '2016p859270', '2016p859378','2016p859625', '2016p859628', '2016p859929', '2016p860053','2016p860215', '2016p860224', '2016p860287', '2016p860567','2016p860729', '2016p861632', '2019p809993', '2019p927023','2020p201743', '2020p475488', '2020p606562'

salichon, 2021-10-05:
- thanks for this description:

- from the 27 events ID perspectives  and the Period of time  identified Strong Motion default in quality **2016.054 till 2017.218.** 
It looks like  22 events may fall into the questionable period of the SM instrument issue.

- the 5 others `'2019p809993', '2019p927023','2020p201743', '2020p475488', '2020p606562'` should have better data set with the strong motion

From the Broadband side of the process: MQZ broadband had a fairly stable sensor and datalogger  since 2009 to now:
`Guralp Systems Ltd.,CMG-3ESPC,T35861,MQZ,10 ,2009-08-06T04:00:00Z,9999-01-01T00:00:00Z` 
 `Kinemetrics,Q330/6,3386,Glencoe Station,,2009-08-06T00:00:01Z,9999-01-01T00:00:00Z`
 
 
 let s focus on the most recent and 5  events first.

salichon, 2021-10-05:
MQZ is located in _Christchurch_

- https://www.geonet.org.nz/earthquake/technical/2019p809993 M5
- https://www.geonet.org.nz/earthquake/technical/2019p927023 M5.4
are located to distance ~700/800km _upper north island_ 
Strong motion tool does not provide SM data. This is legitimate due to signal to  noise ratio too low for relevant result  afaik
https://strongmotion.geonet.org.nz/#/event/2019p927023
https://strongmotion.geonet.org.nz/#/event/2019p809993 

-https://www.geonet.org.nz/earthquake/technical/2020p475488  M5.9  is located in _Fjordland_ ~ ~400 KM  from MQZ
Strong motion tool does  provide SM data https://strongmotion.geonet.org.nz/#/event/2020p475488 

-https://www.geonet.org.nz/earthquake/technical/2020p606562 M5.4  is located in _Fjordland_ ~ ~400 KM  from MQZ
Strong motion tool does  provide SM data https://strongmotion.geonet.org.nz/#/event/2020p606562

- https://www.geonet.org.nz/earthquake/technical/2020p201743  M5.1 is located in _Canterbury_ about 100km 
Strong motion tool does  provide SM data  https://strongmotion.geonet.org.nz/#/event/2020p201743 

Out of 5,  the 3 latests and closest have some reasonable data to compare with for a start

salichon, 2021-10-05:
@SquirrelKnight  In order to focus on the most relevant data sets:
 What are the certain distance criteria for using the data :
 "All stations for which an event arrival was detected and within a certain distance defined by a magnitude-distance scale are used to download waveforms." ? 
 I find that the "certain distance"  can be pretty loose for some distant events and a resulting acceleration can be pretty low for M5 > 500km distance     
 thanks :)

SquirrelKnight, 2021-10-05:
[image]

Here is an image of the magnitude-distance relationship. As you can see, stations selected for lower magnitude data are selected from considerably smaller distances.

I can certainly provide the raw PSA data. Where would be a good place to upload it?

salichon, 2021-10-05:
thanks a lot @SquirrelKnight  
"I can certainly provide the raw PSA data. Where would be a good place to upload it?"  here on git (as *.text) so that we some psa to compare :) 

okay thanks for the distance relationship  !
still  for having  MQZ used in far north 2019p809993 M5 and 2019p927023 M5.4 
I reckon that the selection should be tighter  the epicenter distance does not seem right. 

cheers 
j

SquirrelKnight, 2021-10-05:
As I mentioned above, I do compute IMs for detected arrivals as well. We will be providing a pared down version of the database that excises any IMs outside of this distance relationship, but I've made it an objective to supply as much data as possible! Looking at 2019p927023, the signal appears quite strong (these figures are both unfiltered with no instrument correction).

[image]

[image]

SquirrelKnight, 2021-10-05:
I wanted to add that I have also produced similar plots, but in consideration of the hypocentral distance. Here is an example for MQZ:
[image]

closed 2021-10-12 by salichon

SquirrelKnight, 2021-10-18:
> @SquirrelKnight <https://github.com/SquirrelKnight>
> first iteration of change for the station MQZ strong:
> 1- Legacy site code change (20 to 21) (location of instrument)
> 2-Correction of the orientation at least from Mid 2019 to oreination of N
> to azimuth 270
>
> *Warning: Just noticed that it may contain a bug into the metadata built
> for historical data: orientation of the strong is considered 270 deg since
> the start of the Strong motio)* as in
>
> https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=MQZ&level=channel&format=text&channel=HN
> ?
>
> Second stage will be : fix the metadata and rename the data component
> properly in the 12Z system
>
> FYI @elidana <https://github.com/elidana>

Great. Thanks for digging in deep!

salichon, 2021-11-07:
So the 2016 orientation is deemed correct ENZ.  from https://github.com/GeoNet/tickets/issues/9655#issuecomment-954285451
from then the assumption is that the cabling/configuration 2016-2019 was perhaps messed up with an install put away since 2019.
So Only testing data will bring a some ideas / and fix.

salichon, 2021-11-15:
For  checking  the above assumption (or similar consequence)
 
-  use event 2020p2020p606562 
-  swap PSA output components (for the sake of simplicity)  
-  Compute ratio 
-  plot respective component ratio **HHE Vs HNN and HHN Vs HNE** 

----------------------------------------------
**Swapped PSA output** 
[image]

As opposed to https://github.com/GeoNet/help/issues/84#issuecomment-940482069
with:
[image]

So several things here (still to be solidified)  @elidana @SquirrelKnight  

1 - HN strong motion Horizontal components are most likely **Inverted** HNE is confirmed actually HNN!  (and HNN is HNE) 
or more precisely HHN aligns much better with HNE in terms of output  (HHE with HNN)
2 - Bearing is verified on the field  and being 270 which is **another Added source of confusion**  for naming channel orientation being likely South and West
3- To be verified: It looks like that the process (strong motion tools Geonet)  I used  to check does not rotate the channels with regards to the azimuth.   
 
 Hence badly named and orientated  component chanels are a nightmare to deal with overtime.

salichon, 2021-11-15:
- Checking not the MQZ Strong Season II 2016 Feb to  2019 June 
- Selected events  (from Jesse' s batch) Good SNR. prior to 2016 only veet from Kaikoura aftershocks 14-15Nove (so limited)  
Picked fro try '2016p858147' ( M5 ~100km Monday 14 November) , '9', '2016p861632' (  M5 ~100km Tue Nov 15),

salichon, 2021-11-16:
2017p795065 psa comparison between BN and HH is consistent
[image]
strong assumption the period _2016_2019 seasonII_  is all good in terms  of data/metadata - 
QC  and critical trace inspectio was required

FYi @elidana

salichon, 2021-11-16:
--- Summary ---- 
MQZ HH Vs Strong motion  (2016-2020)

- HH is all good and properly orientated 

- SM  motion parameters derived from either  instruments are okay and consistent in values, consistent with the GeoNet tools. 

- Discrepancies for the period 2019/2020 is thought to be related to SM Horizontal channel naming  and their  orientation with regards to the weak motion. (basically inverted channels ) - Very confusing.

- Many of the processing do not account for Azimuth or do not account for it when geographical coordinates.

- SM from 2016 to 2019 is looking like properly named and orientated 

- Discrepancies shown for  the set of events in 2016 is thought to be related to the choice of data within an immediate aftershock sequences. SNR + coda + superimposed events + local site effects (2 different locations)  + instrumentation 
Even though the data looks really good quality for both instruments at that time. it looks like some frequency content differences. 
as in 
[image]

-Data from 2017 with similar instrumentation do not show major difference between both instrumentations derived parameters.
As seen from the derived parameters.

  
 FYI @elidana @SquirrelKnight @mnaguit et al ...

elidana, 2021-11-16:
summary

broadband data need to be corrected:
- from 2019xx : do we need to update channel names or swap components? rename channels to HN1/HN2 would be preferable and more clear
- possibly visit the site and rerotate the sensor (or check sensor config and cables)

- if using BN HN : all good (@SquirrelKnight can you please add your latest graph for this station?) - at high frequency difference might be Nyquist..

salichon, 2022-02-22:
@SquirrelKnight @elidana @mnaguit 
MQZ strong was reorientated,
as you can check in https://service.geonet.org.nz/fdsnws/station/1/query?station=MQZ&level=channel&format=text&channel=HNE,HNN
there is a long legacy of instrumentation orientation for this site, the later:
_NZ|MQZ|21|HNN|Azimuth |270.000000_

Has been modified on the field ~10:30 on 23/02/2022 NZT. (and will be updated in Delta asap)

- N and E as they should be  -  Someone to verify ?
- We will task/work back the legacy from  the period at least:  2016-02-24 to Feb 2022
 on the data and the metadata side of the thing - this will need a bit of  time and caution 

upon completion close this painful  ticket!! :) 

cheers
jerome