GeoNet/help #85: RIZ issue with broadband vs. accelerometer comparison

elidana, 2021-10-01:
Jesse Hutchinson (University of Canterbury) has identified an issue with RIZ seismic data when comparing data from the broadband and strong motion sensors installed at the station.

His procedure is explained in https://github.com/GeoNet/help/issues/84#issuecomment-931798699

processed Power Spectral Amplitude plots show a consistent difference between RINZ strong motion and broadband data for earthquakes with M>5 in the time period 2019-2020 
[image]

It is likely that this difference is due to one of the sensors going flat during that time period. Further investigations to confirm the cause are needed.

/cc @SquirrelKnight, to help pinpoint time period and possible cause, can you please post here the list of events used to produce this figure?

SquirrelKnight, 2021-10-01:
Here are the events used for the above measurements:

'2012p553582', '2012p595069', '2012p647755', '2012p681111',
       '2012p733490', '2012p766923', '2012p767122', '2020p263458',
       '2020p269386', '2020p332789', '2020p489837', '2020p498617',
       '2020p525251', '2020p548309', '2020p830671', '2020p926977',
       '2020p979274'

salichon, 2021-11-22:
RIZ is Raoul island Remoteley accessble stations 

- considering metadata HH Vs HN as in  https://service.geonet.org.nz/fdsnws/station/1/query?station=RIZ&format=text&level=channel&channel=H?Z

- Season Events 2012 is consistent and all good with  no instrumentation changes
 
- Season Event 2020's is more complicated - metadata changes over time _indicate potential issue(s)_   
 a. **Sep 2019 to Nov 2020** for the events 
`'2020p263458','2020p269386', '2020p332789', '2020p489837', '2020p498617','2020p525251', '2020p548309'`
 b. **Nov2020 to  Feb 2021** for the events  `'2020p830671', '2020p926977','2020p979274'`

salichon, 2021-11-22:
- most likely  a ratio of 4  between HH and HN for these 2 above events.   (more to come)

salichon, 2021-11-22:
There is an - obvious-  PSA/PGA difference in 2020  by RIZ Raoul island between HH/HN  See belpw (no need to compare the PSAs  Ratio is ~time 4 

Event 2020p498617

```
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>         <?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<!DOCTYPE earthquake SYSTEM "stationlist.dtd">                  <!DOCTYPE earthquake SYSTEM "stationlist.dtd">
<stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"   <stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"
  <station code="RIZ" name="RIZ" netid="NZ" insttype="Trilliu |   <station code="RIZ" name="RIZ" netid="NZ" insttype="FBA-ES-
    <comp name="10.HHE">                                      |     <comp name="20.HNE">
      <acc value="3.2882998009" flag="0"/>                    |       <acc value="0.7794464103" flag="0"/>
      <vel value="1.2413733920" flag="0"/>                    |       <vel value="0.3010102158" flag="0"/>
      <psa03 value="5.2516239164" flag="0"/>                  |       <psa03 value="1.3026295004" flag="0"/>
      <psa10 value="0.5281061817" flag="0"/>                  |       <psa10 value="0.1306030042" flag="0"/>
      <psa30 value="0.0349679485" flag="0"/>                  |       <psa30 value="0.0086528750" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHN">                                      |     <comp name="20.HNN">
      <acc value="5.3235685639" flag="0"/>                    |       <acc value="1.4449198090" flag="0"/>
      <vel value="1.5349515077" flag="0"/>                    |       <vel value="0.3551027246" flag="0"/>
      <psa03 value="5.6611978434" flag="0"/>                  |       <psa03 value="1.4561612866" flag="0"/>
      <psa10 value="0.3342218803" flag="0"/>                  |       <psa10 value="0.0880081106" flag="0"/>
      <psa30 value="0.0292224447" flag="0"/>                  |       <psa30 value="0.0076309110" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHZ">                                      |     <comp name="20.HNZ">
      <acc value="3.5350992197" flag="0"/>                    |       <acc value="0.4533266631" flag="0"/>
      <vel value="0.6877658960" flag="0"/>                    |       <vel value="0.1281977146" flag="0"/>
      <psa03 value="1.2282583205" flag="0"/>                  |       <psa03 value="0.2997003001" flag="0"/>
      <psa10 value="0.1445904646" flag="0"/>                  |       <psa10 value="0.0345638400" flag="0"/>
      <psa30 value="0.0115920347" flag="0"/>                  |       <psa30 value="0.0028616972" flag="0"/>
    </comp>                                                         </comp>
  </station>                                                      </station>
</stationlist>                                                  </stationlist>
```
and
event  2020p979274 

> <?xml version="1.0" encoding="UTF-8" standalone="yes"?>         <?xml version="1.0" encoding="UTF-8" standalone="yes"?>
> <!DOCTYPE earthquake SYSTEM "stationlist.dtd">                  <!DOCTYPE earthquake SYSTEM "stationlist.dtd">
> <stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"   <stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"
>   <station code="RIZ" name="RIZ" netid="NZ" insttype="Trilliu |   <station code="RIZ" name="RIZ" netid="NZ" insttype="FBA-ES-
>     <comp name="10.HHE">                                      |     <comp name="20.HNE">
>       <acc value="0.4941437803" flag="0"/>                    |       <acc value="0.1170454285" flag="0"/>
>       <vel value="0.1115322176" flag="0"/>                    |       <vel value="0.0282150623" flag="0"/>
>       <psa03 value="0.4723697473" flag="0"/>                  |       <psa03 value="0.1200957546" flag="0"/>
>       <psa10 value="0.0250678201" flag="0"/>                  |       <psa10 value="0.0062046699" flag="0"/>
>       <psa30 value="0.0022416171" flag="0"/>                  |       <psa30 value="0.0005787207" flag="0"/>
>     </comp>                                                         </comp>
>     <comp name="10.HHN">                                      |     <comp name="20.HNN">
>       <acc value="0.3657536328" flag="0"/>                    |       <acc value="0.0811721977" flag="0"/>
>       <vel value="0.0933475966" flag="0"/>                    |       <vel value="0.0241417859" flag="0"/>
>       <psa03 value="0.5231511377" flag="0"/>                  |       <psa03 value="0.1355195799" flag="0"/>
>       <psa10 value="0.0209135042" flag="0"/>                  |       <psa10 value="0.0053807146" flag="0"/>
>       <psa30 value="0.0019890443" flag="0"/>                  |       <psa30 value="0.0005128434" flag="0"/>
>     </comp>                                                         </comp>
>     <comp name="10.HHZ">                                      |     <comp name="20.HNZ">
>       <acc value="0.2302177579" flag="0"/>                    |       <acc value="0.0649731152" flag="0"/>
>       <vel value="0.0510248189" flag="0"/>                    |       <vel value="0.0134712057" flag="0"/>
>       <psa03 value="0.2288046422" flag="0"/>                  |       <psa03 value="0.0575988433" flag="0"/>
>       <psa10 value="0.0118311645" flag="0"/>                  |       <psa10 value="0.0029701454" flag="0"/>
>       <psa30 value="0.0008189927" flag="0"/>                  |       <psa30 value="0.0002130991" flag="0"/>
>     </comp>                                                         </comp>
>   </station>                                                      </station>

salichon, 2021-11-23:
- comparison of Ratio HH/BN(HN)   good up to 2016 (checked on  2016p475517 and 2016p424618) 
Events from  2017 (2017p258746 and 2018p531317)  start to look awkyard   in terms of values matching
-Assumption: 
This can be related to a datalogger swap between instrumentations resulting into  something bizarre after that ? 
check https://github.com/GeoNet/tickets/issues/882

FYI @SquirrelKnight @elidana  WIP

salichon, 2021-11-23:
- from _2018 July_ RIZ strong motion degraded noticeably  up to 2018.261/263 (September) where the Strong motion became useless up  to **2019.264** _Sept 21_  where the measurement became sensible
- So 
Period of really Bad bad RIZ strong motion  **2018.261/263**  till **2019.264**

salichon, 2021-11-23:
- Change happened in 2016 September Raoul mission (2016.260)
- Actual proper recording of the strong motion start from **2016.266**
-  BN/HN 20 RIZ  2016.265  is crap (~weak motion recordings instead of strong) and is corrected from 2016.266
[image]

salichon, 2021-11-24:
- Checking SM and parameters  for the period late 2016 Strong motion such as **2016p872630. November 2016 M5.**4
**RIZ SM Vs Weak Parameters show  a noticeable difference - just after a field change** 
 Please check: https://github.com/GeoNet/tickets/issues/879-  
 
  
  as in:
 ```
`  <station code="GLKZ" name="GLKZ" netid="NZ" insttype="Trill     <station code="GLKZ" name="GLKZ" netid="NZ" insttype="Trill
    <comp name="10.HHE">                                            <comp name="10.HHE">
      <acc value="0.8105100364" flag="0"/>                            <acc value="0.8105100364" flag="0"/>
      <vel value="0.3625093726" flag="0"/>                            <vel value="0.3625093726" flag="0"/>
      <psa03 value="2.0332093732" flag="0"/>                          <psa03 value="2.0332093732" flag="0"/>
      <psa10 value="0.3249875917" flag="0"/>                          <psa10 value="0.3249875917" flag="0"/>
      <psa30 value="0.0213950094" flag="0"/>                          <psa30 value="0.0213950094" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHN">                                            <comp name="10.HHN">
      <acc value="1.0078450475" flag="0"/>                            <acc value="1.0078450475" flag="0"/>
      <vel value="0.2934706799" flag="0"/>                            <vel value="0.2934706799" flag="0"/>
      <psa03 value="2.2630825714" flag="0"/>                          <psa03 value="2.2630825714" flag="0"/>
      <psa10 value="0.3567158499" flag="0"/>                          <psa10 value="0.3567158499" flag="0"/>
      <psa30 value="0.0251648408" flag="0"/>                          <psa30 value="0.0251648408" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHZ">                                            <comp name="10.HHZ">
      <acc value="0.5360655152" flag="0"/>                            <acc value="0.5360655152" flag="0"/>
      <vel value="0.2156726317" flag="0"/>                            <vel value="0.2156726317" flag="0"/>
      <psa03 value="1.1372011097" flag="0"/>                          <psa03 value="1.1372011097" flag="0"/>
      <psa10 value="0.1286487524" flag="0"/>                          <psa10 value="0.1286487524" flag="0"/>
      <psa30 value="0.0169301900" flag="0"/>                          <psa30 value="0.0169301900" flag="0"/>
    </comp>                                                         </comp>
  </station>                                                      </station>
  <station code="RIZ" name="RIZ" netid="NZ" insttype="CMG-3ES |   <station code="RIZ" name="RIZ" netid="NZ" insttype="FBA-ES-
    <comp name="10.HHE">                                      |     <comp name="20.BNE">
      <acc value="1.7598043392" flag="0"/>                    |       <acc value="0.4247914995" flag="0"/>
      <vel value="0.6878929891" flag="0"/>                    |       <vel value="0.1668835032" flag="0"/>
      <psa03 value="3.6848863627" flag="0"/>                  |       <psa03 value="0.9317593622" flag="0"/>
      <psa10 value="0.5529704634" flag="0"/>                  |       <psa10 value="0.1386621672" flag="0"/>
      <psa30 value="0.0628844313" flag="0"/>                  |       <psa30 value="0.0164746375" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHN">                                      |     <comp name="20.BNN">
      <acc value="1.4987779734" flag="0"/>                    |       <acc value="0.4084177596" flag="0"/>
      <vel value="0.4324893035" flag="0"/>                    |       <vel value="0.1096524787" flag="0"/>
      <psa03 value="3.4138858626" flag="0"/>                  |       <psa03 value="0.9107063442" flag="0"/>
      <psa10 value="0.3917531014" flag="0"/>                  |       <psa10 value="0.1067686554" flag="0"/>
      <psa30 value="0.0550700525" flag="0"/>                  |       <psa30 value="0.0137542494" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHZ">                                      |     <comp name="20.BNZ">
      <acc value="0.8925252623" flag="0"/>                    |       <acc value="0.2127553540" flag="0"/>
      <vel value="0.2790970367" flag="0"/>                    |       <vel value="0.0671373821" flag="0"/>
      <psa03 value="1.3901107669" flag="0"/>                  |       <psa03 value="0.3529664395" flag="0"/>
      <psa10 value="0.2443403190" flag="0"/>                  |       <psa10 value="0.0627944020" flag="0"/>
      <psa30 value="0.0250792215" flag="0"/>                  |       <psa30 value="0.0064768656" flag="0"/>
    </comp>                                                         </comp>
  </station>                                                      </station>
`
```
[image]

and a consistent 1/4 ratio between HH and BN on RIZ until a n strong motion issue along 2018 until a replacement in 2019 sept 
FYI @SquirrelKnight @elidana @ozym 

--- 
Need to check if the HH Vs BN/HN ratio is still 4  in 2021

salichon, 2021-11-24:
results for **2021p715493 M5.5 **  **GLKZ  and RIZ consistent output ~1%g  Vs RIZ strong motion  0.3%g**  
<stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"   <stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"
  <station code="GLKZ" name="GLKZ" netid="NZ" insttype="Trill     <station code="GLKZ" name="GLKZ" netid="NZ" insttype="Trill
    <comp name="10.HHE">                                            <comp name="10.HHE">
      <acc value="1.2434342470" flag="0"/>                            <acc value="1.2434342470" flag="0"/>
      <vel value="0.3226558428" flag="0"/>                            <vel value="0.3226558428" flag="0"/>
      <psa03 value="1.1799961255" flag="0"/>                          <psa03 value="1.1799961255" flag="0"/>
      <psa10 value="0.1873236385" flag="0"/>                          <psa10 value="0.1873236385" flag="0"/>
      <psa30 value="0.0122821212" flag="0"/>                          <psa30 value="0.0122821212" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHN">                                            <comp name="10.HHN">
      <acc value="1.0757444905" flag="0"/>                            <acc value="1.0757444905" flag="0"/>
      <vel value="0.2776559241" flag="0"/>                            <vel value="0.2776559241" flag="0"/>
      <psa03 value="1.1076307169" flag="0"/>                          <psa03 value="1.1076307169" flag="0"/>
      <psa10 value="0.1566439131" flag="0"/>                          <psa10 value="0.1566439131" flag="0"/>
      <psa30 value="0.0104541844" flag="0"/>                          <psa30 value="0.0104541844" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHZ">                                            <comp name="10.HHZ">
      <acc value="1.0311887968" flag="0"/>                            <acc value="1.0311887968" flag="0"/>
      <vel value="0.1782112942" flag="0"/>                            <vel value="0.1782112942" flag="0"/>
      <psa03 value="0.7133046028" flag="0"/>                          <psa03 value="0.7133046028" flag="0"/>
      <psa10 value="0.0506322085" flag="0"/>                          <psa10 value="0.0506322085" flag="0"/>
      <psa30 value="0.0068765641" flag="0"/>                          <psa30 value="0.0068765641" flag="0"/>
    </comp>                                                         </comp>
  </station>                                                      </station>
  <station code="RIZ" name="RIZ" netid="NZ" insttype="Trilliu |   <station code="RIZ" name="RIZ" netid="NZ" insttype="FBA-ES-
    <comp name="10.HHE">                                      |     <comp name="20.HNE">
      <acc value="1.5805146737" flag="0"/>                    |       <acc value="0.3822916356" flag="0"/>
      <vel value="0.6992911070" flag="0"/>                    |       <vel value="0.1721719665" flag="0"/>
      <psa03 value="3.9067508958" flag="0"/>                  |       <psa03 value="0.9890861015" flag="0"/>
      <psa10 value="0.3648450089" flag="0"/>                  |       <psa10 value="0.0904114417" flag="0"/>
      <psa30 value="0.0254664582" flag="0"/>                  |       <psa30 value="0.0064493938" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHN">                                      |     <comp name="20.HNN">
      <acc value="2.5137795811" flag="0"/>                    |       <acc value="0.5764419711" flag="0"/>
      <vel value="0.6970251287" flag="0"/>                    |       <vel value="0.1708662387" flag="0"/>
      <psa03 value="2.7988747423" flag="0"/>                  |       <psa03 value="0.7049039446" flag="0"/>
      <psa10 value="0.1880126050" flag="0"/>                  |       <psa10 value="0.0498220628" flag="0"/>
      <psa30 value="0.0116712685" flag="0"/>                  |       <psa30 value="0.0030854582" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHZ">                                      |     <comp name="20.HNZ">
      <acc value="1.1997534487" flag="0"/>                    |       <acc value="0.3079534936" flag="0"/>
      <vel value="0.2771769896" flag="0"/>                    |       <vel value="0.0701502819" flag="0"/>
      <psa03 value="1.4687793185" flag="0"/>                  |       <psa03 value="0.3726367741" flag="0"/>
      <psa10 value="0.0740232073" flag="0"/>                  |       <psa10 value="0.0187514684" flag="0"/>
      <psa30 value="0.0093899779" flag="0"/>                  |       <psa30 value="0.0023705178" flag="0"/>
    </comp>                                                         </comp>
  </station>                                                      </station>
</stationlist>                                                  </stationlist>

salichon, 2021-11-24:
- Potential legacy RIZ SM 2016 to Now (2021) Sept  debatable measures output 
- Actions when all confirmed with other sources : 
         - to reconfigure and fix the SM from "Now" (remotely or on the field)
         - to  amend the past data 2016-"Now" either with the data or the metadata 
         - To document this for later
 Note: in addition to RIZ and GLKS  BB to compare strong motion parameters      , MMKS strong is a new station on the island since **2021-10-24**

FYI @elidana @mnaguit  

Perhaps to be  communicated  into a delta meeting

salichon, 2021-11-24:
Summary  most likely scenarios (today) 
- Season1:  RIZ strong from 2012 (and most likely before)  to 2016  September  OK 
- Season2:  Sep 2016  to ~ 2018.261 waveform Can be used but requires caution on the measures
- Season3:  Sep 2018 to  2019.264 Say Kaputt - not usable 
- Season4:  Sep2019  to  2021 and TBC  Can be used but requires caution on the measures
- Spin-off: from Oct2021 MMKS strong can be used.

salichon, 2022-02-01:
- [ ] testing with a response with Obsidian4x fix on 2021p876265

salichon, 2022-02-16:
@SquirrelKnight 
This instrument related issue will be addressed soon while patching the metadata.
it will encompass a series of sites as per https://github.com/GeoNet/tickets/issues/9943
over identified period of times. (related to Obsidian datalogger 4X response) 

cheers 
FYI  @elidana @mnaguit

SquirrelKnight, 2022-02-16:
> it will encompass a series of sites as per GeoNet/tickets#9943

That is wonderful to hear. Thank you for all of the hard work!

salichon, 2022-02-17:
PR was merged correction applied today @SquirrelKnight 
documentaion to follow up with the sites affected

salichon, 2022-02-17:
- Former overall sensitivity for the RIZ SM streams as
 1.70934447112192×10⁰⁶ counts/m.s-2
-  New/updated is ~ 4.3e+5  counts/m.s-2

Re-open if additional issues

closed 2022-02-17 by salichon