GeoNet/help #86: FOZ issue with broadband vs. accelerometer comparison

SquirrelKnight, 2021-10-01:
Similar to station RIZ, FOZ shows a time-dependent issue, most likely due to flattening of one of the two sensors.
[image]

The events used for these computations are as follows:

'2020p111984', '2020p201743', '2020p475488', '2020p606562',
       '2802756', '2808300', '2829153', '2853545'

elidana, 2021-10-17:
our team had a look at FOZ installation history, below their findings.
Since 2004 the same broadband (CMG-3ESP) and strong motion (FBA-ES-T) sensors have been installed and always co-located in a vault side by side. In 2015 the vault was moved. Orientations of the installed sensors in each vault epoch looks all good (see pictures below). 

@SquirrelKnight can you please confirm what is the event that does not show an homogeneous difference across the spectrum?
(should be one from the group '2802756', '2808300', '2829153', '2853545')

#### 2004
[image]

#### 2015
[image]

salichon, 2021-11-29:
from above "FOZ - sugarloafquarry is causing me a bit of confusion as a 4X is currently present there since 2017 (= I expect a problem) yet Jesse's plots for 2020 events shows a 1:1 ratio "   (--> thought to be the opposite) 

- Actions follow up with this and a list of events between 2008 and now. 
to confirm the reposne and exclude other potential additional (BB and or SM) contribution affecting the measures. and the derived SM parameters

salichon, 2021-11-29:
FOZ metadata history: https://service.geonet.org.nz/fdsnws/station/1/query?station=FOZ&format=text&level=channel&channel=H?Z

```
#Network | Station | Location | Channel |  Elevation | Azimuth | Dip | SensorDescription | Scale | ScaleFreq | ScaleUnits | SampleRate | StartTime | EndTime
NZ|FOZ|10|HHZ|0.000000|0.000000|-90.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2004-10-13T00:00:00|2015-09-22T18:00:00
NZ|FOZ|20|HNZ|0.000000|0.000000|-90.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2004-10-13T00:00:02|2015-09-24T23:59:59
NZ|FOZ|11|HHZ|0.000000|0.000000|-90.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2015-09-23T05:30:00|2017-03-14T21:20:00
NZ|FOZ|21|HNZ|0.000000|0.000000|-90.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2015-09-25T00:00:00|2017-03-14T21:20:00
NZ|FOZ|21|HNZ|0.000000|0.000000|-90.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2017-03-14T21:20:01|2017-03-15T23:59:59
NZ|FOZ|11|HHZ|0.000000|0.000000|-90.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2017-03-14T21:20:01|2018-12-11T21:00:00
NZ|FOZ|21|HNZ|-0.000000|0.000000|-90.000000|Accelerometer|1709344.471122|1.000000|m/s**2|200.000000|2017-03-16T00:00:01|
NZ|FOZ|11|HHZ|0.000000|0.000000|-90.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2018-12-11T21:45:00|
```

- Some data maybe be stored under '10' or '20' though can be described as '11' or '21' (soemthingto check) over time  since 2015

salichon, 2021-11-29:
events to be looked at in more detaills 
- season 1  2008 M5.5 285...,xml 2007 M5.7 28083....xml 
- season 2 2020 M5.9 2020p475488 and  M5.4 2020p606562 

Eachs season has SM and BB set up differentl
y

salichon, 2021-11-29:
1. season 1 (only 2008 event) 
PGA/PGV values very consistent from one instrum,ent to another: 

```
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>         <?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<!DOCTYPE earthquake SYSTEM "stationlist.dtd">                  <!DOCTYPE earthquake SYSTEM "stationlist.dtd">
<stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"   <stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"
  <station code="FOZ" name="FOZ" netid="NZ" insttype="CMG-3ES |   <station code="FOZ" name="FOZ" netid="NZ" insttype="FBA-ES-
    <comp name="10.HHE">                                      |     <comp name="20.BNE">
      <acc value="0.0323674290" flag="0"/>                    |       <acc value="0.0327546128" flag="0"/>
      <vel value="0.0402220914" flag="0"/>                    |       <vel value="0.0406699982" flag="0"/>
      <psa03 value="0.0851590240" flag="0"/>                  |       <psa03 value="0.0758072721" flag="0"/>
      <psa10 value="0.0393508916" flag="0"/>                  |       <psa10 value="0.0386904403" flag="0"/>
      <psa30 value="0.0139519885" flag="0"/>                  |       <psa30 value="0.0139745425" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHN">                                      |     <comp name="20.BNN">
      <acc value="0.0471940290" flag="0"/>                    |       <acc value="0.0468957958" flag="0"/>
      <vel value="0.0311889867" flag="0"/>                    |       <vel value="0.0319635127" flag="0"/>
      <psa03 value="0.1137600603" flag="0"/>                  |       <psa03 value="0.1137505145" flag="0"/>
      <psa10 value="0.0324292424" flag="0"/>                  |       <psa10 value="0.0323447519" flag="0"/>
      <psa30 value="0.0082503008" flag="0"/>                  |       <psa30 value="0.0082962589" flag="0"/>
    </comp>                                                         </comp>
    <comp name="10.HHZ">                                      |     <comp name="20.BNZ">
      <acc value="0.0262786899" flag="0"/>                    |       <acc value="0.0267345734" flag="0"/>
      <vel value="0.0356911684" flag="0"/>                    |       <vel value="0.0357763512" flag="0"/>
      <psa03 value="0.0484587420" flag="0"/>                  |       <psa03 value="0.0479322373" flag="0"/>
      <psa10 value="0.0395912395" flag="0"/>                  |       <psa10 value="0.0399410097" flag="0"/>
      <psa30 value="0.0134465935" flag="0"/>                  |       <psa30 value="0.0134981134" flag="0"/>
    </comp>                                                         </comp>
  </station>                                                      </station>
</stationlist>                                                  </stationlist>
```

salichon, 2021-11-30:
- 2020p475488 M5.9 
Obvious difference! 0.25 ratio as for other OBSIDIAN4x  
```
<stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"   <stationlist created="" xmlns="ch.ethz.sed.shakemap.usgs.xml"
  <station code="FOZ" name="FOZ" netid="NZ" insttype="CMG-3ES |   <station code="FOZ" name="FOZ" netid="NZ" insttype="FBA-ES-
    <comp name="11.HHE">                                      |     <comp name="21.HNE">
      <acc value="0.1964913459" flag="0"/>                    |       <acc value="0.0490636793" flag="0"/>
      <vel value="0.0911897759" flag="0"/>                    |       <vel value="0.0231956877" flag="0"/>
      <psa03 value="0.4294555146" flag="0"/>                  |       <psa03 value="0.1096788029" flag="0"/>
      <psa10 value="0.1006291272" flag="0"/>                  |       <psa10 value="0.0261726216" flag="0"/>
      <psa30 value="0.0282471822" flag="0"/>                  |       <psa30 value="0.0073772619" flag="0"/>
    </comp>                                                         </comp>
    <comp name="11.HHN">                                      |     <comp name="21.HNN">
      <acc value="0.1927286157" flag="0"/>                    |       <acc value="0.0486891813" flag="0"/>
      <vel value="0.0979078193" flag="0"/>                    |       <vel value="0.0248455348" flag="0"/>
      <psa03 value="0.3392446491" flag="0"/>                  |       <psa03 value="0.0890793055" flag="0"/>
      <psa10 value="0.1049995760" flag="0"/>                  |       <psa10 value="0.0262968559" flag="0"/>
      <psa30 value="0.0146964616" flag="0"/>                  |       <psa30 value="0.0035166214" flag="0"/>
    </comp>                                                         </comp>
    <comp name="11.HHZ">                                      |     <comp name="21.HNZ">
      <acc value="0.0760153330" flag="0"/>                    |       <acc value="0.0187925837" flag="0"/>
      <vel value="0.0641462353" flag="0"/>                    |       <vel value="0.0165146793" flag="0"/>
      <psa03 value="0.1482640661" flag="0"/>                  |       <psa03 value="0.0380991274" flag="0"/>
      <psa10 value="0.0845038442" flag="0"/>                  |       <psa10 value="0.0215602451" flag="0"/>
      <psa30 value="0.0224075067" flag="0"/>                  |       <psa30 value="0.0057510434" flag="0"/>
```
- Same for 2020p606562
```
 <station code="FOZ" name="FOZ" netid="NZ" insttype="CMG-3ES |   <station code="FOZ" name="FOZ" netid="NZ" insttype="FBA-ES-
    <comp name="11.HHE">                                      |     <comp name="21.HNE">
      <acc value="0.1533250202" flag="0"/>                    |       <acc value="0.0393349780" flag="0"/>
      <vel value="0.0898074424" flag="0"/>                    |       <vel value="0.0227910726" flag="0"/>
      <psa03 value="0.4868539400" flag="0"/>                  |       <psa03 value="0.1219592245" flag="0"/>
      <psa10 value="0.0769432494" flag="0"/>                  |       <psa10 value="0.0198753541" flag="0"/>
      <psa30 value="0.0107528644" flag="0"/>                  |       <psa30 value="0.0028317628" flag="0"/>
    </comp>                                                         </comp>
    <comp name="11.HHN">                                      |     <comp name="21.HNN">
      <acc value="0.1347125750" flag="0"/>                    |       <acc value="0.0329017764" flag="0"/>
      <vel value="0.0654058228" flag="0"/>                    |       <vel value="0.0168343461" flag="0"/>
      <psa03 value="0.5619861549" flag="0"/>                  |       <psa03 value="0.1477659948" flag="0"/>
      <psa10 value="0.0487740774" flag="0"/>                  |       <psa10 value="0.0123426318" flag="0"/>
      <psa30 value="0.0054993729" flag="0"/>                  |       <psa30 value="0.0014363258" flag="0"/>
    </comp>                                                         </comp>
    <comp name="11.HHZ">                                      |     <comp name="21.HNZ">
      <acc value="0.0749384243" flag="0"/>                    |       <acc value="0.0186263893" flag="0"/>
      <vel value="0.0426370664" flag="0"/>                    |       <vel value="0.0108029659" flag="0"/>
      <psa03 value="0.2159569157" flag="0"/>                  |       <psa03 value="0.0552651060" flag="0"/>
      <psa10 value="0.0404837977" flag="0"/>                  |       <psa10 value="0.0103638821" flag="0"/>
      <psa30 value="0.0062699189" flag="0"/>                  |       <psa30 value="0.0016113766" flag="0"/>
    </comp>                                                         </comp>

```

salichon, 2021-11-30:
@elidana @SquirrelKnight 
relates to https://github.com/GeoNet/help/issues/85

And points all the aqttention to the `Datalogger Obsidian4x` response or internal configuration

elidana, 2021-12-14:
https://github.com/GeoNet/delta/pull/1234

salichon, 2022-02-17:
is addressed through Obsidian4X respons
e afaik and will be documented in https://github.com/GeoNet/help/issues/94
@SquirrelKnight  any other concerns about FOZ ?

closed 2022-02-21 by salichon