GeoNet/help #87: KNZ issue with broadband vs. accelerometer comparison

SquirrelKnight, 2021-10-01:
Possible time-dependent issue with station KNZ.

[image]

Events used for computations are as follows:

'2014p422047', '2014p751490', '2015p718332', '2020p239907',
       '2020p263458', '2020p598678'

elidana, 2021-10-04:
thanks a lot for providing those details @SquirrelKnight !

Can you please confirm what is the event (within the group of 2020) that is giving the mismatch? Thanks!

salichon, 2021-11-30:
From:  https://service.geonet.org.nz/fdsnws/station/1/query?station=KNZ&format=text&level=channel&channel=HNZ 
and information above
with a `ln` ratio of `0.7` that implies atgh the PGA values are roughly 1/2 half (or twice)  of their actual values

salichon, 2021-11-30:
**PSA HH Vs  HN are consistent from 2021Sep : 2021p665417 event** 
[image]

Although i might recommend to check the orienataiton of _KNZ HN 21_  (one additional close events. 

---------------------------
KNZ HH Vs HH  summary  (strong likelihood for) 
**Instrumentation  2019/2021 issue fixed in June 2021** 

FYI @SquirrelKnight @elidana 

Ig happy we can close this one :) 

most likey using  a nearby/sizable ebevent to confirm this point (the above close is 4.9 only)

salichon, 2022-02-17:
- [ ] Check f anyother work about this one ?

salichon, 2022-03-15:
- test on 2019p738432 M>5 Oct 01   dist <100km  - period 2019/07  to 2020/06
    traces are looking okay (BN)  No triggered HN though 
    
- [ ] check HH/BN comparison values  with the corrected response

salichon, 2022-03-17:
@mnaguit the KNZ ground is 
`KNZ | Kokohu | Stable |   | On concrete floor of purpose-built instrument vault | Near base of steep rock spur | Hikurangi Subduction Zone, Raukumara F2 (c. 11 km) |   | A, Rock (Vs greater than 600m/s) or very thin soil (less than 5m) over rock | B, Rock | A, Rock or very stiff soil 
`
So  to me there are no reason for rasonable site amplification at the time. 

@mnaguit i would aim at getting final - peer -  confirmation that  the SM observations at this national site  are questionable for the period 2019/07 to 2021/06.

Cheers :)

closed 2023-09-25 by salichon