GeoNet/help #91: Broadband vs. accelerometer comparison, bulk

SquirrelKnight, 2021-11-03:
Hi, I have been working on pulling together a zip document that compiles the previously shared comparisons. These files include figures showing pSA differences coloured according to datetime, magnitude and rupture distance, PGA differences, and direct waveform comparisons. The file structure is organized by magnitude -> comparison channels -> figure type. Currently I have generated comparisons between HH and HN channels for M ranges of 4.5 - 5 and 5 - 6. I have also included CSV files with the intensity measure data for channel pairs, as well as a CSV with mean PSA differences and standard deviations for each station.

Here is a Dropbox link for the data: https://www.dropbox.com/s/30ix5mn28a80hnb/gm_compare.zip?dl=0

Here is an example of the new waveform comparison figures. Note that the data for both instrument types are displayed in acceleration units of m/s**2. Seismometer data was converted to acceleration using previously described pre-processing techniques.
[image]

I hope that this will prove to be a valuable resource. I plan to update this in the future with alternative channel comparisons and M 4 - 4.5 data.

elidana, 2021-11-05:
Thanks a lot @SquirrelKnight ! We have downloaded your new results, and those will surely be valuable.

elidana, 2021-11-16:
suggest to use
score_X_HN	f_min_X_HN	score_Y_HN	f_min_Y_HN	score_Z_HN	f_min_Z_HN	score_X_diff	score_Y_diff	score_Z_diff
in `psa_data.csv`

salichon, 2021-11-30:
- Need to identify and split the station of interest

salichon, 2022-03-06:
- [ ] SYZ  SMvsBB  response corrected 2022 
     Although one event comparison has a ratio that does not fit exactly the comparison. (see figure https://github.com/GeoNet/tickets/issues/9943#issuecomment-979451235)  ratio is ~ -6 and -4
    that is as i understand   _e-4 and e-6 reps. 1/50  or 1/400 times the value expected  between bot sensor !_   
    provided the updated response correction  this should be "now" something in the order  resp 1/12.5 and  1/100
    **Yet still big difference!** 
      - Assumption of a cumulative problem with the response.  (BB installation or sensors ?) within  the studied time frame.
     **Requires data  and id of events for further investigation** 
     Confirmation Ticket to be created for enquiry please
     FYI @elidana @mnaguit @SquirrelKnight

salichon, 2022-03-16:
- [ ] Finalized KNZ story  https://github.com/GeoNet/help/issues/87

salichon, 2022-03-16:
next batch worth to look at and identify potenetial issue 

- [ ] DSZ
- [ ] MWZ
- [ ] MRZ
- [ ] MLZ
- [ ] PUZ
- [ ] URZ
- [ ] WVZ

salichon, 2022-03-16:
another batch

- [ ] BFZ 
- [ ] HAZ
- [ ] HIZ 
- [ ] JCZ
- [ ] MQZ
- [ ] MXZ
- [ ] OPZ
- [ ] OUZ
- [ ] RPZ
- [ ] TLZ
- [ ] TOZ

From there based upon the inspection weshoudl have  fixed or explained most of the questioning.
assuming that few instrumentation failure/inconsistencies may still be present on the overall batch over limited time periods. Would need addtional detailed inspection .

FYI @elidana @mnaguit