GeoNet/help #92: CTZ response information not available prior to 2021-11-24

calum-chamberlain, 2021-12-01:
As of last week, response information for CTZ appears to be missing from prior to 2021-11-24T19:37:00.000000Z as noted in [this PR](https://github.com/eqcorrscan/EQcorrscan/pull/473).

This can be observed in the [webrequest here](https://service.geonet.org.nz/fdsnws/station/1/query?station=CTZ&level=response).

ozym, 2021-12-01:
@salichon looks like the `Broadband` role for

- connections.csv
`CTZ,10,Wharekauri,Broadband,2007-12-15T07:00:00Z,9999-01-01T00:00:00Z`

hasn't been updated for the dataloggers (okay for current one, but not for old ones as per above date range).
- dataloggers.csv
 `Kinemetrics,Q330HR/6,2258,Wharekauri,,2007-12-15T06:00:01Z,2021-11-23T03:00:00Z`

This needs the "Broadband" bit added.

salichon, 2021-12-02:
- will be fixed soon @calum-chamberlain 
FYI @elidana

calum-chamberlain, 2021-12-02:
Thanks! Feel free to close this whenever is useful for you.

elidana, 2021-12-02:
metadata fixed in GeoNet/delta#1225

closed 2021-12-02 by salichon