GeoNet/help #100: MRZ - Strong Motion Vs BB - comparison enquiry

salichon, 2022-03-21:
- waveforms derived from instruments have differences
 -   could be polarities for older events
 -   Can be  amplitudes for events 2020 

- [x]  List of events of interest 
polarity check evid 2498995 and 231214
amplitudes **2020p802823** **2020p464185**
2020p434252 2020p877108 2020p947812 2020p972758
- [ ]  Isolate if new issues arise and are confirmed
- [ ] Propose solution / fix /Actions
- [ ] Document

salichon, 2023-09-25:
- [ ] Check station change history SM and BB if available  
- [ ] Compare waveforms Horizontals and Vertical in velocity and acceleration for given reference events and periods 
- [ ] Check PGA values for events of reference over time periods for both instruments
- [ ] Check additional waveform parameters 

- [ ] Check if all compenents are consistent (Horizontal Vs Vertical)

mnaguit, 2023-09-26:
2020 M4.7 Levin ([2020p802823](https://www.geonet.org.nz/earthquake/2020p802823))

- PGA estimates
```
PGA (HNE) = 0.019081, PGA (HNN) = 0.028269, PGA (HNZ) = 0.012876 m/s/s 
PGA (HNE) = 0.001946, PGA (HNN) = 0.002883, PGA (HNZ) = 0.001313 g 
PGA (HHE) = -0.019427, PGA (HHN) = 0.011166, PGA (HHZ) = 0.009958 m/s/s 
PGA (HHE) = -0.001981, PGA (HHN) = 0.001139, PGA (HHZ) = 0.001015 g 
```

- Acc subplots
[image]

- Vel subplots
[image]

mnaguit, 2023-09-26:
**2020 M4.6 Whakatane ([2020p464185](https://www.geonet.org.nz/earthquake/2020p464185))**

- PGA estimates
```
PGA (HNE) = -0.003669, PGA (HNN) = 0.003887, PGA (HNZ) = 0.001751 m/s/s 
PGA (HNE) = -0.000374, PGA (HNN) = 0.000396, PGA (HNZ) = 0.000179 g 
PGA (HHE) = 0.003156, PGA (HHN) = -0.002764, PGA (HHZ) = 0.001142 m/s/s 
PGA (HHE) = 0.000322, PGA (HHN) = -0.000282, PGA (HHZ) = 0.000116 g 
```

- Acc subplots
[image]

- Vel subplots
[image]

mnaguit, 2023-10-02:
Discussion summary (as of `Oct 2, 2023`)
- polarity issue confirmed, all components may be reversed from `2003 to 2011` (this is captured in `polarities.csv`, which is good!)
- waveforms corrected to ACC & VEL look comparable, along with peak motions using two events in 2020 (but cannot generalize based on 2 events!)
- cannot finalize an outcome just depending on selected events/scenarios
  - what happens for longer epicentral distances?
  - how both sensors perform over a longer range of time, not just event-based?
  - wider discussion on polarities may include like delta vs xml, or the extent by which to 'red flag' stations like this other than in `polarities.csv` which may not probably give heads up to users not familiar with delta)  
- this assessment is linked with a data user query. the events were selected by the data user. 
- most selected events have short epicentral distances (lower than 60km) : data user files in https://www.dropbox.com/s/30ix5mn28a80hnb/gm_compare.zip?dl=0

salichon, 2023-10-02:
@Thomas-Benson @mnaguit 
Even better about the `"polarities.csv"` of MRZ these polarities are well accounted for 
(we may need to review the definition of polarities (i forgot that bit)
 - For the period 2003-2011:
    The SM sensor has an Azimuth to 0  in delta
    The polarity change is accounted for in the  `"polarities.csv"`  
    The final results  is https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=MRZ&level=channel&format=text&channel=HN?
    and gets the right components orientation 

Thanks @Thomas-Benson for pointing that out ! well done - a No PB for that period.

Thomas-Benson, 2023-10-05:
This doesn't really contribute that much, but I was testing some scripts for PSA calculations and plotting and wanted to put something here.
Example with [2019p927023](https://www.geonet.org.nz/earthquake/2021p169083) which Jerome has done above:
[image]

Testing with the March 2021 7.2 East Cape quake 2021p169083 at distance 524Km for a slighly larger epicentral distance.
Acceleration waveforms look OK:
[image]
PSA plots:
[image]
[image]
[image]

I get good PSA values for both the Strong motion and broadband sensors, which also matches up fairly well with the values available on the [strong motion page](https://strongmotion.geonet.org.nz/#/event/2021p169083) for the event 
| SM | BB | GeoNet strong motion tool (HN) |
| ----- | ----- | ---- |
| HNE PSA0.3: PSA 0.151885 | HHE PSA0.3: PSA 0.166159 | 0.152314
HNE PSA1.0: PSA 0.03727 | HHE PSA1.0: PSA 0.041462 |  0.0391874
HNE PSA3.0: PSA 0.006473 |  HHE PSA3.0: PSA 0.006867 | 0.00626706
HNN PSA0.3: PSA 0.162689 | HHN PSA0.3: PSA 0.172862 | 0.17351
HNN PSA1.0: PSA 0.045646 | HHN PSA1.0: PSA 0.039204 | 0.0474634
HNN PSA3.0: PSA 0.006985 | HHN PSA3.0: PSA 0.006272 | 0.00659691
HNZ PSA0.3: PSA 0.059772 | HHZ PSA0.3: PSA 0.064563 | 0.0629257
HNZ PSA1.0: PSA 0.023733 | HHZ PSA1.0: PSA 0.022504 | 0.0272351
HNZ PSA3.0: PSA 0.004603 | HHZ PSA3.0: PSA 0.004533 | 0.00459017

mnaguit, 2023-10-10:
As of `10 October 2023`

- last item to check: SM vs BB spectral acc ratios being too symmetric (we can close this ticket after this last check! 👍 )
- to do next: replicate these steps, choose another station (https://github.com/GeoNet/help/issues/91#issuecomment-1068699372)
- final verdict in terms of data quality 
  - SM vs BB channel components have comparable processed peak motions
  - wrt selected events, both sensors are working as expected (noting that the performance of these sensors at MRZ was checked at certain time periods only)

closed 2023-10-17 by mnaguit