GeoNet/help #101: Stations WQSZ WMSZ WPSZ WDSZ WTSZ - Alpine Temp experiment  - data metadata hosted by IRIS/Geonet -actions

salichon, 2022-06-07:
Place holder to have a discussion with VUW @calum-chamberlain  et al. 
- All but WQSZ are hosted by IRIS 
- Data are stored at Geonet (at least a fair bunch).
- Metadata are not totally complete  (response) 

Context and information required prior to action to complete or amend the GeoNet repos.

1- Is WQSZ part of the 6F experiment and should it be transfered?
2- When all the data/meta  are complete agree about the local repository  and actions at Geonet wrt to tmp.  experiments. 

cheers

calum-chamberlain, 2022-07-11:
1. I believe that WQSZ was renamed from [WHAT2](http://ds.iris.edu/mda/9F/WHAT2/?starttime=2009-03-22&endtime=2020-10-26) by GeoNet, but we (and IRIS) have retained the WHAT2 name.

Not sure I can answer 2! Thanks for thinking about this Jerome.

salichon, 2022-07-12:
Kia Ora @calum-chamberlain 
 from the IRIS  side of the thing :  https://service.iris.edu/fdsnws/station/1/query?net=6F&level=station&format=text 
 no  `WHAT2`  in the 6F but the 9F network 

a bit of a case. 
 
  in delta : _WQSZ,SP,Whataroa Quarry,-43.2792,170.3604,96,,WGS84,2014-06-23T00:00:05Z,9999-01-01T00:00:00Z_ 
  (and still opened)
  WHAT2 9F was closed in 2020.  
 
 This WQSZ/WHAT2 is something  that we shall work on to get the right references and information  for data and metadata to one or the other DMCs

calum-chamberlain, 2022-07-12:
Kia ora Jerome, yes that is right WHAT2 was always part of the SAMBA network (9F), and has been since ~2008. We have not been allowed to set an open end-date in IRIS for our "temporary" networks, but we retain the SAMBA (9F) sites. We plan on updating those data and metadata at IRIS in the next few weeks while we archive as much of our VUW hosted data as we can.

I would close the WQSZ station at the last date that you have data for it. We will archive data from that site to IRIS, not to GeoNet.

staylorofford, 2022-08-17:
Hi @calum-chamberlain,

I am looking to complete our sensor orientation dataset and WMSZ, WPSZ, WTSZ, WDSZ all have unknown orientations in our metadata.

In the simplest case, we could update the orientations for these (if known, or calculate them if not).

However, if the data are hosted by IRIS and the metadata are also complete with IRIS or another third party, it is necessary to ask whether GeoNet should have these data (and for WQSZ) in its data/metadata archive. I appreciate this is a heavy question, but do these data _need_ to be in GeoNet's archive, or can they be accessed elsewhere by all those who would need to access them?

To add complexity, if GeoNet _owns_ the data at all, and it is not entirely VUW's, then this question becomes moot and we will want to complete the data/metadata archive for the sites. Ergo, my first question...

calum-chamberlain, 2022-08-22:
Kia ora @staylorofford - I think that the orientations of these sensors are now known thanks to some of the work that John Townend emailed you about. Hopefully we can find out more from Carolin Boese when she gets back to John.

I'm unsure about what data are available where, and it would be best to do a check of data availability between IRIS and GeoNet. We do not plan on archiving more data from these stations to GeoNet though, and would put the data at the IRIS DMC. 

In terms of ownership, that question would have to go to John who oversaw the deployment. As far as I know only the comms system during real-time monitoring was owned by GeoNet.

salichon, 2025-02-02:
**IRIS** 
_Metadata_
WQSZ is WHAT2 on 9F 
9F|WHAT2|-43.278999|170.360001|95.0|X|2009-03-22T00:00:00.0000|2020-10-26T00:00:00.0000 

Only WHAT not WHAT2 has metadata available.

_Metadata_
Other station on the network  6F 
6F|WDSZ|-43.293659|170.412354|97.4|Whataroa|2013-02-02T11:59:59.0000|2020-10-26T00:00:00.0000
6F|WMSZ|-43.280876|170.395493|87.6|Mint Creek, Whataroa|2013-02-02T11:59:59.0000|2020-10-26T00:00:00.0000
6F|WPSZ|-43.283707|170.41394|95.5|Parker Creek, Whataroa|2013-02-02T11:59:59.0000|2020-10-26T00:00:00.0000
6F|WTSZ|-43.301762|170.412354|105.2|Tommys Creek, Whataroa|2013-02-02T11:59:59.0000|2020-10-26T00:00:00.0000

Response available: https://service.iris.edu/fdsnws/station/1/query?net=6F&level=response&station=W?SZ 

- WBSZ (NZ/..) is _not refered_ by IRIS either in networks 9F or 6F  and needs investigation 
as in 
WBSZ,SP,Whataroa River Valley,-43.290821,170.406577,70,,WGS84,2015-10-01T00:00:00Z,99namenl99-01-01T00:00:00Z

------

**GeoNet** 

_Metadata_
- **only WHSZ is informed with complete metadata**  - Network NZ 
https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=W?SZ&level=channel&format=text

 - Other stations are not detailed beyond their location in delta WQSZ **WMSZ WPSZ WDSZ WTSZ  WBSZ

_Data_
```
WDSZ:  2014.246 - 2017.243
WMSZ: 2014.246 - 2017.245
WPSZ:  2024.246 - 2017.283
WQSZ: 2014.246 - 2017.304 
WTSZ:  2014.189 - 2017.243

```
NO ** WBSZ** data No **WHAT2** 

----

Actions: 
 
- [ ] Collect **WDSZ  WMSZ WPSZ WTSZ** Iris 6F metadata and update delata (check data) 
- [ ] Contact JohnTowned/Callum for instrument response WHAT2/WQSZ ? 
- [ ] Data/metadata correction alignements to perform and WBSZ to be cleared from Delta ?  

FYI  and actions @Thomas-Benson  :)