GeoNet/help #102: FDSN CLIENT ISSUE WITH GETTING mseed file

melahmad, 2022-07-06:
Hello,

So i am trying to get a miniSeed or Seed file having 3 component wave form data using the URL query.

I tried doing this and its not working, can someone please let me know whats wrong. The link is pasted below.

https://service.geonet.org.nz/fdsnws/dataselect/1/query?network=NZ&station=HPSC&channel=BN?&starttime=2011-02-21T00:00:00.000&endtime=2011-02-22T00:00:00.000

On another note i tried downloading the data of this event as CSV file from the website. Is there a way to transfer these sheets to miniSeed file.

Please help.

Best,

salichon, 2022-07-06:
Hello @melahmad 
HSPC is a strobng motion 
You you check which instrumentaion and channels are avaible on a station for instance usign the station service:
https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=HPSC&level=channel&format=text 

You ll see that HPSC is a strong motion 200Hz with a `HN?` streams.

Cheers!
Jerome

salichon, 2022-07-06:
Hi @melahmad 
I don t know how work Geospy,
I would advise to refer to their manual to work your issues 

Meanwhile the data provided have 3 components as per example above.

also with regards to the data  services i woudl  striongly advise to have a look at the tutorials:
https://github.com/GeoNet/fdsn/tree/main/examples 

best regards
Jerome 
FYI @mnaguit

mnaguit, 2022-07-06:
From your question posted thru email (_what do you exactly meant by 12Z ? vertical components only?_ )

This is a convention used under channel code. 12Z represents a three-component sensor: Z for the vertical component and 12 for the 2 horizontal components. 
Full details on stream naming conventions in this [link](https://www.geonet.org.nz/data/supplementary/channels#:~:text=The%20current%20National%20Strong%20Motion,a%20last%20being%20a%20Z%20.) 

FYI @melahmad

salichon, 2022-07-15:
i reckno this is solved at our ends - unless any addtional question or issues ;) 
cheers

melahmad, 2022-07-17:
Thank you both.

closed 2022-07-17 by melahmad