GeoNet/help #104: SCTS showing glitches

segburg, 2023-03-09:
Kia ora,
I just discovered that my data processing is failing because of Secretary Island strong motion station SCTS. SCTS waveform data regularly shows glitches, gaps and overlaps. The attached screenshots show clear glitches on HNE and HNZ before 1am. Interestingly the glitches don't happen at the same time on all components. The glitch is followed by a series of gaps and one overlap. The second figure shows a glitch in detail. Overall I have also found the data collected at this station to be particularly noisy.
[image]
[image]

staylorofford, 2023-03-09:
Hi @segburg, this looks like a tricky one.

SCTS is definitely one of the noisest sites in our network, but what you show is anomalous. It is possible that it relates to lightning, as that area is home to a great deal of it. Do you have a few examples of when you see this signal which we could investigate?

Regarding gaps and overlaps, that may be a communications issue. I'll table that with our network engineers and see what they have to say. A site like SCTS might simply be in too extreme an environment to guarantee a high completeness of data, but even then there shouldn't be corruption like you mention.

segburg, 2023-03-10:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe:

I haven't looked at all the data but have caught similar glitches on
2021/11/05
2021/12/03
2022/03/06
2022/03/08

I have so far only focussed on the period from 2021/11 to 2022/03. The glitches don't seem to happen too often for that period of time.

Cheers
Sandra

Notice: This email and any attachments are confidential and may not be used, published or redistributed without the prior written consent of the Institute of Geological and Nuclear Sciences Limited (GNS Science). If received in error please destroy and immediately notify GNS Science. Do not copy or disclose the contents.

elidana, 2025-08-26:
bumped into this one and there was an update we forgot to post here.

Looking at some internal conversation, looks like those "glitches" might be attributed to nearby lightning strikes.

@segburg thanks a lot for bringing this to our attention!

closed 2025-08-26 by elidana