GeoNet/help #106: MGCS -  shift Z channel time series

ElenaManea, 2023-03-15:
Hi,
 
can you please check the channels times for this station - MGCS? it looks like a time shift for Z chan.
Eventid: 2016p859929, 2016-11-14 04:10:21.

Cheers,
Elena

[image]

[image]

salichon, 2023-03-15:
thanks @ElenaManea 
With regards  to https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=MGCS&&format=text&level=channel
- the instrumentation has not changed since 2015  and  HN is avalble in contnous since 2020. 
- All looks good in delta:  https://github.com/GeoNet/delta/search?q=MGCS

to me
- Ealier event this day do not show time difference between channels. see below 
- Channels are times with the same GPS and datalogger so most likely good or bad at the same time   

[image]

So far i don t see any issue so 
the assumption is then  
-  202611 14 is period of aftershocks hence please ensure you re associating the same events (Some(and only some  are localized see https://service.geonet.org.nz/fdsnws/event/1/query?starttime=2016-11-14T04:00:00&endtime=2016-11-14T04:20:00&format=text)   
and nmaybe  try by selecting only events disambiguated from superimposed ongoing event waveforms  
(perhaps outside of intense  aftershocks sequence )
 
Or 
-  i don t understand the issue completely  and need more info :) 

Cheers
jerome

salichon, 2023-03-17:
- [ ] I reckon  timing is looking okay
@mnaguit  what do you think   :) ?

mnaguit, 2023-03-17:
From quick checking, timing looks fine for all channels.

mnaguit, 2023-03-17:
Further details / routine check on data-metadata alignment (although @salichon has already touched on most of these points):
- earthquake event: `https://www.geonet.org.nz/earthquake/technical/2016p859929` 
   - two streams of data are available for this event: `BN(12Z)` & `HN(12Z) `
- sensor history: from `FBA Etna` to `CUSP3D` - sensor swap happened in 2015 (`https://service.geonet.org.nz/fdsnws/station/1/query?sta=MGCS&format=text&level=channel&channel=*`)
- data plots/products:
   - instrumented corrected plots, consisted with plots of @ElenaManea (except for the channel component naming - `BN(ENZ)` vs `BN(12Z)` --just an observation, i'm not sure if that affects any of your processing particularly the sensor orientation?)
   - if processed products help, data products for this station (unfiltered & filtered time series, response spectra & summary of peak ground motions are available here: `https://strongmotion.geonet.org.nz/#/event/2016p859929`)

[image]

salichon, 2023-03-19:
@ElenaManea unless you d have some more insights  around the timing 
we ll close this thread soon :)

salichon, 2023-03-19:
done done done

closed 2023-03-19 by salichon