GeoNet/help #110: GLKZ Instrument Respons for HH channels missing prior to 2021-10-24

SquirrelKnight, 2023-06-23:
Hello, I noticed that instrument responses for the HH channels are missing for station GLKZ prior to 2021-10-24 from the Obspy FDSN webservice. Please see the attached image. I believe this information was previously available, at least as of August 2022.

[image]

salichon, 2023-06-25:
Hi @SquirrelKnight good to hear from you :) 
https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=GLKZ&level=channel&format=text&channel=?H? 
is confirmed
the metadata volume for GLKZ weak motion is broken - this gonna be fixed
thanks for noticing

SquirrelKnight, 2023-06-26:
Thanks @salichon ! Good to hear from you too :) Glad to hear that the fix is on the way!

salichon, 2023-09-14:
- [ ]  check for :
https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=GLKZ&level=channel&format=text&channel=?H? 
upon recent change in the GLKZ metadata

salichon, 2024-03-12:
- merged by March 2024 -should be fixed

closed 2024-03-12 by salichon