GeoNet/help #112: LHZ seismic station polarity issue

elidana, 2023-07-21:
A user of our seismic data notified us about the following issue, so opening this as it might be useful for others

copying here the user request, and will add some initial info as a first comment

> Dear station operator， 
> I am a seismology researcher in China. I recently processed LHZ data from the NZ network in New Zealand for the period 2008-2012 and found that the waveform records from several stations had polarity flips. The waveform data and instrument response files we used were downloaded from IRIS. 
> In the measurement of the relative traveltime of the 10 mHz surface wave, it is found that all the measured values of the problem station have a half-period (50 s) deviation for several years. Since our measurement method is based on a cross-correlation technique, it is presumed that the cause is flipped polarity of the waveform recording. Afterwards, we used the P wave of the teleseismic (Wenchuan M7.9 earthquake in China on May 12, 2008, and the Tokyo M9.1 earthquake in Japan on March 11, 2011) to conduct a test. Compared with other nearby normal stations (IU SNZO, NZ RPZ), the seismic records of station NZ BKZ on the island of New Zealand have opposite polarities. For details, please refer to the attachment.
> I recently plan to publish an article about station clock error and polarity reversal, and I hope the station operator can tell me the reason for the polarity reversal of the waveform record of this station. I will thank you in the acknowledgments of the article. 
> 
> 

> 
> 
> Earthquakes and station locations picked for this example
>  
> [image]
> 
> 
> 
> The first arrival P waves recorded by the three stations, raw data is unprocessed except for filtering (0.01Hz-0.02Hz). The record in the middle shows a clear polarity reversal
>   
> [image]
> 
> [image]
>

elidana, 2023-07-21:
Unfortunately the reverse polarity on some of the GeoNet broadband stations is a known issue (see https://github.com/GeoNet/help/issues/36 for another example).

When known, reversed polarities are tracked into our metadata repository (https://doi.org/10.21420/0VY2-C144).
The info can be found in the following table: https://github.com/GeoNet/delta/blob/main/install/streams.csv
and when `Reversed` value is set to `true`, it means that 

For this specific request BKZ station is a known issue, 

Station|Location|Band|Source|Sampling Rate|Axial|Reversed|Triggered|Start Date|End Date
------|--------|----|------|-------- ----|-----|--------|---------|----- ----|--- ----
BKZ|10|V|H|0.1|false|true|false|2004-02-11T02:00:00Z|2013-10-09T00:00:00Z
BKZ|10|V|H|0.1|false|false|false|2013-10-09T00:00:01Z|2013-12-31T00:00:00Z
BKZ|10|L|H|1|false|true|false|2004-02-11T02:00:00Z|2013-10-09T00:00:00Z
BKZ|10|L|H|1|false|false|false|2013-10-09T00:00:01Z|9999-01-01T00:00:00Z
BKZ|10|H|H|100|false|true|false|2004-02-11T02:00:00Z|2013-10-09T00:00:00Z
BKZ|10|H|H|100|false|false|false|2013-10-09T00:00:01Z|9999-01-01T00:00:00Z
BKZ|20|B|N|50|false|false|false|2013-10-09T00:00:01Z|2020-06-16T00:00:00Z
BKZ|20|H|N|200|false|false|true|2004-02-11T02:00:00Z|2020-01-31T03:30:00Z
BKZ|21|H|N|200|false|false|false|2021-10-22T00:32:00Z|9999-01-01T00:00:00Z

shun-zz, 2023-07-24:
Hi, elidana. Thank you for your reply. I compared my test results with the list you provided. Most of the problem stations are consistent. But one station, QRZ, showed polarity reversal in my detection(My detection is only for the LHZ component). But in the list you gave it is normal. I used the teleseismic P waves of earthquakes with a magnitude greater than 7 for a test, and found that the polarity reversal did occur at this station.

In my detection results, the first period of polarity reversal at QRZ station: 232 days before 2008, and the second period: from the 290th day in 2008 to the 220th day in 2009. These timings coincided well with the timing of instrument changes at [station.](https://ds.iris.edu/mda/NZ/QRZ/?starttime=1990-04-26&endtime=2599-12-31)

 I hope the station operator can verify this.

[image]
[image]
[image]
[image]
[image]
[image]

elidana, 2023-07-27:
Dear @shun-zz ,

I have missed another important files that contain polarities information in delta, apologies for that!
This is https://github.com/GeoNet/delta/blob/main/install/polarities.csv

That file is only listing the time periods for which a specific site and channel may have reversed polarity.
QRZ is listed there as a station with reversed polarities for time periods that seems to match with your observations

Station | Location | Sublocation | Subsource | Primary | Reversed | Method | Citation | Start Date | End Date
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
QRZ | 10 |   |   | true | true |   |   | 2003-08-06T00:00:00Z | 2003-10-29T22:00:00Z
113 | QRZ | 10 |   |   | true | true |   |   | 2003-10-30T21:00:00Z | 2007-08-24T09:50:01Z
114 | QRZ | 10 |   |   | true | true |   |   | 2007-08-24T10:00:01Z | 2008-08-24T23:45:00Z
115 | QRZ | 10 |   |   | true | true |   |   | 2008-08-24T23:55:01Z | 2009-08-12T23:00:00Z

We also have confirmations from @jristau analysis for QRZ (John's analysis is also looking at vertical components only), reported below

> QRZ:
> Unknown polarity from 2003-09-25 to 2004-07-15, then reversed from 2004-07-15 to 2009-08-12. Likely reversed the entire time and there were other issues with the station early on which meant I couldn’t check the polarity.

> 
> Station|Start Date|End Date| Reversed(*)
> -- | -- | -- | --
> QRZ | 2003-09-25T19:50:26Z | 2004-07-15T04:27:14Z | U
> QRZ | 2004-07-15T04:27:14Z | 2009-08-12T23:00:00Z | R
> QRZ | 2009-08-12T23:01:01Z | 9999-01-01T00:00:00Z | N
> 
> Reversed: U=Unknown, R=reversed, N=normal

Thanks for providing these further evidences and confirming the already existing informations! That's super helpful

By looking at the station metadata change, there have been a few dataloggers changes during this time period.

Details can be obtaining using our FDSN station metadata service