GeoNet/help #114: Sites missing from GeoNet sensor map

calum-chamberlain, 2023-09-22:
Currently the [GeoNet sensor map](https://www.geonet.org.nz/data/network/sensor/search) is only showing Air Pressure sensors when all station types are selected. I have seen this on multiple computers now over the last two days and at least one of my students is seeing the same.

I am using Firefox (I guess this might be a browser related problem?), and can provide more details as needed.

calum-chamberlain, 2023-09-22:
Sites are also missing for me viewed from Chrome.

mnaguit, 2023-09-22:
Hi @calum-chamberlain , I'm using Chrome. I checked the sensor map, this is the result:
[image]

calum-chamberlain, 2023-09-22:
Possibly a Linux only issue or browser version issue then.  I only get air pressure for chrome and Firefox.

calum-chamberlain, 2023-09-22:
On Firefox for android nothing at all is plotted.

elidana, 2023-09-22:
thanks @calum-chamberlain  for raising it!

agree that it looks like a mobile phone issue (linux browsers works fine for me too, chrome&firefox).

I have the same issue on my iPhone (only air pressure data show)

Model/iOS is: `iPhone SE 16.6.1`

@mnaguit , @calum-chamberlain can you please add your model/version as well? might help with troubleshooting

calum-chamberlain, 2023-09-22:
Firefox 117.0.1 running on Manjaro (Arch) Linux:
[image]

Chrome Version 117.0.5938.88 on Manjaro (Arch) Linux:
[image]

calum-chamberlain, 2023-09-22:
On firefox on Windows (version 102.12.0esr) no sites are plotted.

In case it matters, these are all running on the same (VUW) network.

mnaguit, 2023-09-22:
Same here. No sites plotted using Samsung phone S20+ Android 13 and Pixel Android 10.
I can go ask my hubby to check his phone too. 😆

elidana, 2023-09-22:
odd about the browser @calum-chamberlain ,
I have tried Chrome and Firefox on my Fedora laptop and they both works fine
(I have the last official release of both browsers)

whereas I can replicate your issue on my mobile phone. thanks for adding the sw/version info!

mnaguit, 2023-09-22:
Hi @calum-chamberlain, can you please check if the sensor map is working now on your end? thanks!

calum-chamberlain, 2023-09-22:
Yup, looks good for me now! Thanks

For my own curiosity, did you work out the cause?

sue-h-gns, 2023-09-22:
there was a change to metadata deployed that exposed an issue in the code that failed to deal with null values

closed 2023-10-05 by mnaguit