GeoNet/help #115: CORS headers aren't set properly for FDSN web services

rauwitt, 2023-09-22:
FDSN web services of GEONET can't be accessed by javascript applications.  The following error occurs in the browser console:`
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://service.geonet.org.nz/fdsnws/dataselect/1/version. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing). Status code: 200.` It would be an enhancement to allow FDSN browser access for dynamic web pages in javascript.

elidana, 2023-10-18:
Dear @rauwitt ,

apologies for the belated reply.

Unfortunately, to be compliant with IT security requirements, we cannot implement such an enhancement to the GeoNet FDSN service right now, and without the risk of impacting other users and downstream processes. 

Our FDSN service currently disallows the creation of a webpage which uses javascript to load data.

An hopefully quick question: have you tried if you have the same issue with the FDSN real time service (`service-nrt.geonet.org.nz`)

you would only find the past 7 days of data but the backend of our fdsn-nrt is different from the fdsn-archive service you tried to use. I'm not 100% sure if that will work, but perhaps worth a try?

Yours is a use case we want to consider and keep in mind for future developments, so thanks a lot for reaching out and if you have time we'd be happy to collect your requirements and user story!

rauwitt, 2023-10-22:
Hello elidana,

thanks for considering to enable javascript access as an enhancement for future development.

The error messages appear when trying to use the FDSN event service of GEONET with the URL: service-nrt.geonet.org.nz.
Since the event service is very powerful and allows virtually all kind of queries, it is a pity that you can not use it with javascript.

Currently the following data centers who offer FDNS webserices have implemented javascript access:

1. Geoscience Australia
2. Earthquakes Canada (GOVERNMENT of Canada)
3. USGS (United States Geological Survey)
4. EMSC (European-Mediterranean Seismological Centre)
5. GEOFON (GFZ German Research Center for Geosciences)
6. INGV (Istituto Nazionale di Geofisica e Vulcanologia / Italy)
7. IRIS / SAGE (Seismological Facility for the Advancement of Geoscience )
8.  NCEDC (Northern California Earthquake Data Centerh)
9. SCEDC (Southern California Earthquake Data Center)
10. USP (Centro de Sismologia da Universidade de São Paulo)
11.  NOA (National Observatory of Athens 

Six years ago [Philip Crotwell](https://sc.edu/study/colleges_schools/artsandsciences/earth_ocean_and_environment/our_people/directory/crotwell_philip.php) (University of South Carolina) [remarked](https://www.fdsn.org/message-center/thread/486/): "I feel this kind of access probably should be
allowed and would like to encourage making this work as it is a really powerful way to leverage the existing web services to create dynamic web pages." 

An example of those webpages you can find [here](https://nzquakes.com/new/layout-13.html).

rauwitt, 2024-04-07:
Hello eldiana,

as you mentioned right earlier there is a real danger of server overload if GEONET opens FdSN webservice for
javascript applications.

In this matter it is interesting to see what other FDSN data centers have implemented to prevent
this unwanted side effects.

E.g. USGS tolerates 500 requests per 5 minute.
Exceeding the limit causes a temporary blockage of the webservice for the single ip-adress.

Perhaps this information might be useful for GEONET.

elidana, 2024-06-17:
Dear @rauwitt ,

thanks a lot for the additional information about the USGS implementation.

Unfortunately, with current constraints on GeoNet programme and already planned developments we won't be able to implement such a service in the short term. 
Enabling CORS headers as other data centers is a good suggestion, and we can revisit this in the future if the opportunity arise.

closed 2024-06-17 by elidana