GeoNet/help #116: AVSN station response information

calum-chamberlain, 2023-09-25:
Kia ora all, this might be a longshot, but I wondered if there was response information for the AVSN stations before 2007. The stations in particular that I'm interested in are OTAZ, KAAZ, MTAZ, WTAZ and MKAZ.  Waveform data for these stations are available, and there appears to be some response information from 2003 for some of these stations but it is not continuous. 

For example:
```python
from obspy.clients.fdsn import Client
from obspy import UTCDateTime

bulk = [('NZ', sta, "*", "*", UTCDateTime(2005, 3, 1), UTCDateTime(2005, 3, 2)) 
        for sta in ["OTAZ", "KAAZ", "MTAZ", "WTAZ", "MKAZ"]]
st = client.get_waveforms_bulk(bulk)
inv = client.get_stations_bulk(bulk, level="response")
```

The waveforms have all stations with EHZ channels, but the inventory returned only has station OTAZ, which has no channels associated.

This is very similar to some of my previous questions, apologies. I assume that these responses have been lost, but I thought I would check first.

salichon, 2023-09-25:
relates to: 

- [ ] https://github.com/GeoNet/help/issues/73

salichon, 2023-09-25:
- [ ] Explore KAAZ response content in Delta and detail the mouint for the period 1998-2009

salichon, 2024-08-12:
- [ ] investigate data storage and dates
- [ ] match with metadata periods
- [ ] provide context