GeoNet/help #118: MLZ BB Vs SM investigation

salichon, 2023-10-10:
Task/Check list for investigation  

- [ ] interpret the issue reported ( e.g. Amplitude Diff between BB and SM waveforms)  
inspecting figures oper events per staiton as provided par jessie:  in **gm_compare-1\M_5\HN_HH\wf_figs** 
Checked for events of interest (lot of noise and few good data)  

- [ ] isolate periods of questioning for the period mentioned in that ticket 
  - [ ] Check  event 

   - [ ] Check polarities for cusp events as in
     
-------
- [ ] Check station change history SM and BB if available
  - [ ] Isolate periods of change and potential issue with regards to instrumentation    
     Reversed Strong motion   https://github.com/GeoNet/delta/blob/main/install/polarities.csv 
   
-------  
- [ ] Compare Raw waveforms Horizontals and Vertical in velocity and acceleration for given reference events and periods 
- [ ] Compare waveforms removed form instrument reposnse on  Horizontals and Vertical in velocity and acceleration for given reference events and periods 
- [ ] Check if all components are consistent with  each other  (Horizontal Vs Vertical)
- [ ] Check PGA values for events of reference over time periods for both instruments
- [ ] Check additional waveform parameters if required

mnaguit, 2023-10-13:
**Checking these end user files**
**Event 1: https://www.geonet.org.nz/earthquake/technical/2020p938679** 
[image]

**Event 2: https://www.geonet.org.nz/earthquake/technical/2020p140742** 
[image]

mnaguit, 2023-10-13:
**Station metadata history**

**Strong motion**
```
#Network | Station | Location | Channel | Latitude | Longitude | Elevation | Depth | Azimuth | Dip | SensorDescription | Scale | ScaleFreq | ScaleUnits | SampleRate | StartTime | EndTime
NZ|MLZ|20|HNE|-45.366544|168.118407|594.000000|0.000000|270.000000|0.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2003-09-08T00:00:00|2014-03-25T00:00:00
NZ|MLZ|20|HNN|-45.366544|168.118407|594.000000|0.000000|180.000000|0.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2003-09-08T00:00:00|2014-03-25T00:00:00
NZ|MLZ|20|HNZ|-45.366544|168.118407|594.000000|0.000000|0.000000|90.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2003-09-08T00:00:00|2014-03-25T00:00:00
NZ|MLZ|21|HN1|-45.366530|168.118380|587.000000|0.000000|60.000000|0.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2014-03-25T00:05:00|2019-10-31T21:33:00
NZ|MLZ|21|HN2|-45.366530|168.118380|587.000000|0.000000|150.000000|0.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2014-03-25T00:05:00|2019-10-31T21:33:00
NZ|MLZ|21|HNZ|-45.366530|168.118380|587.000000|0.000000|0.000000|-90.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2014-03-25T00:05:00|2019-10-31T21:33:00
NZ|MLZ|21|HNE|-45.366530|168.118380|587.000000|0.000000|90.000000|0.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2019-10-31T21:34:00|
NZ|MLZ|21|HNN|-45.366530|168.118380|587.000000|0.000000|0.000000|0.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2019-10-31T21:34:00|
NZ|MLZ|21|HNZ|-45.366530|168.118380|587.000000|0.000000|0.000000|-90.000000|Accelerometer|427336.117780|1.000000|m/s**2|200.000000|2019-10-31T21:34:00|
```

**Broadband**
```
#Network | Station | Location | Channel | Latitude | Longitude | Elevation | Depth | Azimuth | Dip | SensorDescription | Scale | ScaleFreq | ScaleUnits | SampleRate | StartTime | EndTime
NZ|MLZ|10|HHE|-45.366544|168.118407|594.000000|0.000000|90.000000|0.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2003-09-08T00:00:00|2014-03-25T00:00:00
NZ|MLZ|10|HHN|-45.366544|168.118407|594.000000|0.000000|0.000000|0.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2003-09-08T00:00:00|2014-03-25T00:00:00
NZ|MLZ|10|HHZ|-45.366544|168.118407|594.000000|0.000000|0.000000|-90.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2003-09-08T00:00:00|2014-03-25T00:00:00
NZ|MLZ|10|HHE|-45.366544|168.118407|594.000000|0.000000|90.000000|0.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2014-03-25T00:00:01|2021-06-08T00:40:00
NZ|MLZ|10|HHN|-45.366544|168.118407|594.000000|0.000000|0.000000|0.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2014-03-25T00:00:01|2021-06-08T00:40:00
NZ|MLZ|10|HHZ|-45.366544|168.118407|594.000000|0.000000|0.000000|-90.000000|Broadband Seismometer|838860800.000000|1.000000|m/s|100.000000|2014-03-25T00:00:01|2021-06-08T00:40:00
NZ|MLZ|10|HHE|-45.366544|168.118407|594.000000|0.000000|90.000000|0.000000|Broadband Seismometer|504365056.000000|1.000000|m/s|100.000000|2021-06-08T04:19:00|2022-07-04T17:22:00
NZ|MLZ|10|HHN|-45.366544|168.118407|594.000000|0.000000|0.000000|0.000000|Broadband Seismometer|504365056.000000|1.000000|m/s|100.000000|2021-06-08T04:19:00|2022-07-04T17:22:00
NZ|MLZ|10|HHZ|-45.366544|168.118407|594.000000|0.000000|0.000000|-90.000000|Broadband Seismometer|504365056.000000|1.000000|m/s|100.000000|2021-06-08T04:19:00|2022-07-04T17:22:00
NZ|MLZ|10|HHE|-45.366544|168.118407|594.000000|0.000000|90.000000|0.000000|Broadband Seismometer|481000000.000000|1.000000|m/s|100.000000|2022-07-07T23:30:00|
NZ|MLZ|10|HHN|-45.366544|168.118407|594.000000|0.000000|0.000000|0.000000|Broadband Seismometer|481000000.000000|1.000000|m/s|100.000000|2022-07-07T23:30:00|
NZ|MLZ|10|HHZ|-45.366544|168.118407|594.000000|0.000000|0.000000|-90.000000|Broadband Seismometer|481000000.000000|1.000000|m/s|100.000000|2022-07-07T23:30:00|
```

mnaguit, 2023-10-13:
**2020 M5.6 Outside NZ ([2020p938679](https://www.geonet.org.nz/earthquake/technical/2020p938679))**

_**PGA estimates**_
```
PGA (HNE) = -0.000197, PGA (HNN) = 0.000355, PGA (HNZ) = -0.000356 m/s/s 
PGA (HNE) = -0.000020, PGA (HNN) = 0.000036, PGA (HNZ) = -0.000036 g 
PGA (HHE) = 0.000052, PGA (HHN) = -0.000044, PGA (HHZ) = -0.000023 m/s/s 
PGA (HHE) = 0.000005, PGA (HHN) = -0.000004, PGA (HHZ) = -0.000002 g 
```

_**Acc subplots**_
[image]

_**Vel subplots**_
[image]

mnaguit, 2023-10-13:
**2020 M4.9 outside NZ ([2020p140742](https://www.geonet.org.nz/earthquake/technical/2020p140742))**

_**PGA estimates**_
```
PGA (HNE) = -0.000111, PGA (HNN) = -0.000156, PGA (HNZ) = -0.000086 m/s/s 
PGA (HNE) = -0.000011, PGA (HNN) = -0.000016, PGA (HNZ) = -0.000009 g 
PGA (HHE) = 0.000075, PGA (HHN) = -0.000058, PGA (HHZ) = 0.000068 m/s/s 
PGA (HHE) = 0.000008, PGA (HHN) = -0.000006, PGA (HHZ) = 0.000007 g 

```
**_Acc subplots_**
[image]

**_Vel subplots_**
[image]

GeoGarfias, 2024-07-04:
I would like to know the conclusion of this investigation. Are the amplitudes comparable? Did you remove the instrument response of the strong motion data and integrate it to get the data to velocity? 
Also, it seems that channels HN* had a high water level value applied when the instrument response was removed.

Thomas-Benson, 2024-07-05:
Hi @GeoGarfias,  @mnaguit and @salichon might be able to give a more complete answer than this once they are back in a few weeks time, but in the meantime I've had a look at a slightly clearer event (2020p475488) than the ones used above and the accelration amplitudes look reasonably comparable:
```

HNZ PSA0.3: 0.068504 m/s/s
HNZ PSA1.0: 0.06009 m/s/s
HNZ PSA3.0: 0.006577 m/s/s
HNZ PSA10.0: 0.001156 m/s/s
HNZ PSA15.0: 0.000386 m/s/s

HHZ PSA0.3: 0.067616 m/s/s
HHZ PSA1.0: 0.058467 m/s/s
HHZ PSA3.0: 0.006398 m/s/s
HHZ PSA10.0: 0.001126 m/s/s
HHZ PSA15.0: 0.000361 m/s/s
```
[image]

Looking at some work supplied by one of the NSHM researchers It appears there was some discrepancies in the period from 2014-2019 while one specific strong motion recorder was installed at the site, but this has now been corrected. Looking at this plot and the history of the site there might be an issue it might be an issue with polarity or sensor orientation, which we are aware was a problem with this installation
[image]