GeoNet/help #119: Seismic-products site is down

mswidan, 2024-01-15:
Hi

https://[data.geonet.org.nz](https://data.geonet.org.nz/seismic-products/strong-motion/)/seismic-products/strong-motion/ and all sub sites are down, please check.

I tried two PCs, each using two different browsers but same issue. Most likely server down.

Appreciate your prompt fix.

 Cheers,
Moustafa
[image]

staylorofford, 2024-01-15:
Hi @mswidan,

Thank you for raising this. We are looking into the issue now and will let you know via this ticket when it is resolved.

staylorofford, 2024-01-15:
Hi @mswidan, the site is now back online.

Please comment here if you have any further issues.

Thanks.

closed 2024-01-15 by staylorofford