GeoNet/help #121: how to use GeoNet seismic waveform data from a local storage

elidana, 2024-02-19:
We have received this user request from @segburg (https://github.com/GeoNet/fdsn/issues/244) , who is using an on premise data store  to analyze seismic waveform data from GeoNet stations.

The local store has some corrupted file, but when downloading data directly from the GeoNet FDSN service or open-data archive those files are complete.

Using this ticket to provide some possible options (potentially interesting for other data users) to have an on-premise copy or sync of the GeoNet seismic waveform data for a given time period.

#### option 1 (small volumes of data) - GeoNet FDSN webservice
The GeoNet FDSN service can be used to request small data volumes or when data from the past 7 days are required. 
Instructions on how to access this service are provided on the [GeoNet website FDSN page](https://www.geonet.org.nz/data/access/FDSN)
and some tutorials are available [in the dataselect jupyter data tutorial](https://github.com/GeoNet/data-tutorials/tree/main/FDSN) 

#### option 2 (moderate to large volumes of data) - GeoNet open data bucket
For moderate to large volumes of data, the recommended approach is to copy data from the GeoNet AWS Open data bucket (https://www.geonet.org.nz/data/access/aws).

Details on how waveform miniseed files are organized in the GeoNet Open AWS archive are provided in the [GeoNet data tutorials](https://github.com/GeoNet/data-tutorials/tree/main/AWS_Open_Data#wave-miniseed)
and some initial instructions and introduction to how to interact with it are provided in this [GeoNet data blog](https://www.geonet.org.nz/news/5MWez34jAuItvAnp12MsQ9)

To interact with the GeoNet open data bucket, the aws command line interface shall be installed on a Unix/Linux machine (https://aws.amazon.com/cli/)

Once the asws-cli is installed, to list the content for a specific year or day and station, the command is
`aws s3 ls --no-sign-request s3://geonet-open-data/waveforms/miniseed/2023`
or
`aws s3 ls --no-sign-request s3://geonet-open-data/waveforms/miniseed/2023/2023.031/WTAZ.NZ/`

To copy one file on your local machine (once the `/home/username/tmp` folder has been created, the command is
`aws s3 cp --no-sign-request s3://geonet-open-data/waveforms/miniseed/2023/2023.031/WTAZ.NZ/2023.031.WTAZ.12-HHE.NZ.D /home/username/tmp/.`

To sync an entire day worth of data to your local machine (on the same  `/home/username/tmp` folder)
`aws s3 sync --no-sign-request s3://geonet-open-data/waveforms/miniseed/2023/2023.031/WTAZ.NZ/ /home/username/tmp/.`
and the same can be applied for all stations available for that day with the following command
`aws s3 sync --no-sign-request s3://geonet-open-data/waveforms/miniseed/2023/2023.031/ /home/username/tmp/.`

We can provide more detailed instructions on how to filter specific waveform data, or different options on how to interact with the AWS open data bucket

calum-chamberlain, 2024-02-19:
For any obspy users, I have hacked together a drop-in replacement for `obspy.clients.fdsn.Client` for use with the GeoNet open-data bucket that anyone is welcome to use (and adapt and fix as needed). This is [here](https://github.com/calum-chamberlain/cjc-utilities/blob/main/src/cjc_utilities/get_data/geonet_aws_client.py) - make sure you test it yourself before trusting it!

salichon, 2024-02-21:
As a short @elidana  @segburg

Open Data bucket to Seiscomp data structure emulation
- [ ] Unix environement  - Install S3fs 
- [ ] Mount OpenDataBucket (ODB) locally so  will show  locally `YYYY/YYYY.Day/sta.Net/YYYY.Day.Sta.loc-Cha.Net.D`
- [ ] Emulate locally the ODB to a parallel local Seiscomp Directory Structure (SDS) ` 
  - [ ] Create Per Year/Day/station/Locid/channel  `YYYY/NET/Sta/Cha.D/` 
  - [ ] Symlink  the ODB file in the proper SDS location  Properly
     `ln -s  YYYY/YYYY.Day/sta.Net/YYYY.Day.Sta.loc-Cha.Net.D    net.sta.loc.cha.D.YYYY.Day`

elidana, 2025-04-16:
closing this as there hasn't been much active conversation recently. But any of the contributors, please feel free to reopen if the question wasn't answered!

closed 2025-04-16 by elidana