GeoNet/help #123: mismatch between seismic waveforms on GeoNet FDSN service and AWS Open Data bucket

elidana, 2024-03-01:
Waveform data  in miniseed format are available on the AWS Open Data bucket.
We are progressively syncing data on this new AWS channel (https://www.geonet.org.nz/data/access/aws and https://www.geonet.org.nz/news/5MWez34jAuItvAnp12MsQ9), so there might be a mismatch of what users can retrieve using the FDSN dataselect service (more complete) and the AWS S3 bucket (catching up), and only make available in AWS data which have metadata we believe are accurate. 
This means that not all miniSEED waveform data available via FDSN are available from AWS Open Data. 

Some users might notice this mismatch, so we would like to use this issue to track those, and fix the sync to AWS. 

A first user got in touch about that, copying below the details. 

We will welcome any further input on that!

> I am only using South Island broadbands and 2-3 strong motions in the very south. 
> I found RPZ channels HHE and HHN and FOZ three channels HHE, HHN, HHZ seem to be consistently missing for 2023.

CallumNZ, 2024-03-03:
`RPZ` and `FOZ` are not on the exclude list for Data Sync.

elidana, 2024-03-04:
thanks for the quick check @CallumNZ , I can confirm that all data for RPZ and FOZ match between FDSN and the Open data bucket

@segburg, for 2023
RPZ HH data channels are labelled as HH1, HH2 and HHZ 
FOZ HH data channels are available under the location code 11 (instead of 10)

see also the FDSN station information for [RPZ](https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=RPZ&level=channel&format=text&channel=HH?) and [FOZ](https://service.geonet.org.nz/fdsnws/station/1/query?network=NZ&station=FOZ&level=channel&format=text&channel=HH?)

closed 2025-09-16 by elidana