GeoNet/help #125: RSAM & SSAM thumbnails not updating

KTJacobs, 2024-09-02:
Kia ora 

It seems that the RSAM & SSAM page thumbnails aren't updating regularly (https://www.geonet.org.nz/volcano/rsamssam).    The thumbnails are really useful for doing a quick scan of activity and getting a broader sense of weather impacts etc.    However, all of the thumbnails appear to be from April 8th.   I realized as I went to look at a few intersting sites and they looked much different when the larger indivdual station plot came up.   The realtime data still comes up when you look at an individual station, so processing appears intact.    What is the configured refresh rate on the thumbnails?  It would be great if this was daily or smaller. 

Images below contain computer time stamp for reference.
Thumbnail from page: 
[image]
Individual large plot: 
[image]

Thanks

elidana, 2024-09-03:
hello @KTJacobs ! Thanks a lot for notifying us about this issue!

Thanks to our wonderful engineers, this has now been fixed, and thumbnails are up to date with latest data.

closed 2024-09-03 by elidana

KTJacobs, 2024-09-03:
Awesome!  Thank you!!

KTJacobs, 2024-09-04:
@elidana Now the timestamps on all the RSAM thumbnails and larger images are from July 31st?

elidana, 2024-09-04:
hei thanks @KTJacobs !

it got me as well, but the date you refer to indicates the start of the time window

the plots are up to date and in sync now (the bug was corrected yesterday)

[image]
[image]

KTJacobs, 2024-09-04:
Of course!  Thank you @elidana  !!

elidana, 2024-09-04:
the real thanks go to our master jedi! ;-)

KTJacobs, 2024-09-04:
Many thanks to the Jedi as well!  Great to have these back up to date.