GeoNet/help #126: Limited or No scale information on Eq Technical Tab Plots

KTJacobs, 2024-10-31:
The 'Technical' Tab for individual GeoNet earthquake locations has a range of information including Time and Magnitude Residuals.    

However, the scale and tick marks of the plots often make them useless for interpreting the information.    This manifests in a few main ways listed below.  Given the efforts invested into making this data available on the website, it would be good to make some minor adjustments to make it useable.

1. For earthquakes which only have picks out to ~1 degree distance: 
-The Pick residual plot often only has tick marks at 0 and 1 degrees.   It would be useful if there were additional subdivisions 0.1 or 0.2 to help get a handle on the distance to the closest station and more easily identify the distance of any changes in slope in the residual plots. 
-The Y-axis could also use additional tick marks (0.1/0.2/0.5?).  Ideally most residuals are below 1 second, so ticks marks only at 1s scale make interpretation more difficult.

-The Station Magnitude Plots sometimes contain NO tick marks for reference! (see attached).    These plots seem to start at the distance of the first magnitude pick.   These need to contain at least 2 tick marks for reference, or could be shown at same scale as residual plots. This problem persists for all magnitude plots that the x axis doesn't start at 0 (and isn't labelled) and either no or a single tick mark are shown making it impossible to identify the x-scale length.   
-Additional tick marks on the magnitude (Y-axis, ~0.2 or 0.50) would also be useful.

[image]
https://www.geonet.org.nz/earthquake/technical/2024p818051

2.  For large earthquakes in the NGMC era especially Pick Residuals are shown in one plot that contains ALL residuals at ALL distances.     
-Especially for large events, this can include distances to 8+ degrees, and the plot distance scale largely focuses on un-used picks.   While the unused pick residuals are of some interest a second plot or a scale that clearly shows the time residuals of the USED picks is of significantly more interest.   One of the problems with showing the residuals of unused picks at large distances is that they are often much larger than for the used picks and expanding the Y-scale to include the worst residuals often obscures the already horizontally condensed used picks.   At large distances it is likely that the time residuals mostly represent the misfit between the 1D velocity model and the true structure so where picks exist at smaller distances, these large distance residuals add very little to interpreting the eq location quality.    Two potential changes that could help make these plots useable are a) letting Y-axis scale be set by used picks and indicating where other picks are off the scale (at limit and in a different color).  b) having a second pick residual plot when distances go beyond 1 or 2 degrees that focusses on the distances of the used picks in a more focused way.    (Image below)
-Similarly to point 1, as it is true of all earthquakes, additional horizontal tick marks at a level less than 1 degree (at least for the first 1 or 2 degrees) would be useful.  
-I'm not suggesting a fixed degree limitation, as we need flexibility so we can have useable information for offshore/tonga-kermadec eqs, and to account for inherent elongated geometry of NZ landmass. 

[image]
https://www.geonet.org.nz/earthquake/technical/2018p816466

3. Final thought if changing plots substantially in the future: 
Due to use of 1D model, distance isn't necessarily an isolated or dominant control on processes affecting time residual or magnitude picks.   Plotting the time residuals or magnitudes in an azimuthal plot could help clarify where/whether residuals are influenced by solution/location misfits and where they are dominated by velocity model complications.  Being able to quickly tease out these two factors would help in assessing solution quality and likelihood to change.    These could be added to the blank space in the origin map (top-left/bottom-right).

elidana, 2024-11-03:
thanks @KTJacobs for those suggestions! we will pass them to the team that maintain and develop the GeoNet website

Jeremy-Houltham, 2024-11-03:
Thanks @KTJacobs 
I have written these out as user stories with acceptance criteria to try and determine the problem clearly for the developers.
Could you please have a read and check if i have summed up your problems correctly?

User stories for the above problem.

As a seismic duty officer I want to see 0.1 increments on the x axis when degree distance is ~1 degree so that I can understand the distance to the closest station to identify the distance of any changes in slope in the residual plots.
AC - Add 0.1 mark increments on the x axis on the Pick Residual plots when the degree is ~1.

As a seismic duty officer I want to see 0.2 increments on the y axis of the pick residuals & stations magnitude plot so that I can easily associate the points with a reference on the y axis.
AC - Add 0.2 mark increments on the y axis on the pick residuals & stations magnitude plots.

As a seismic duty officer I want to see a scale on the x axis of the Stations Magnitude plot so I can determine the distance from the station.
AC - Add a scale to the x axis on the Stations Magnitude plot.

As a seismic duty officer I want to see the used picks only on the pick residuals plot so that I can interpreted the used picks at a clearer resolution and scale.
AC a) letting Y-axis scale be set by used picks and indicating where other picks are off the scale (at limit and in a different color). b) having a second pick residual plot when distances go beyond 1 or 2 degrees that focusses on the distances of the used picks in a more focused way. (Image below)

As a seismic duty officer I want to see the time residuals or magnitudes in an azimuthal plot so that I can clarify if the residuals are influenced by solution/location misfits and where they are dominated by velocity model complications in order to assess solution quality and if they are likely to change.
AC - Add an azimuthal plot for residual picks on the EQ technical page.

Jeremy-Houltham, 2025-02-03:
I need some validation from @KTJacobs if the user stories and acceptance criteria are correct from what you have experienced.