GeoNet/help #127: parsing times in FDSN station request

crotwell, 2024-11-21:
Although not part of the official FDSN web service spec, many data centers allow the appending of a `Z` to a time to indicate UTC, and I feel this is a useful feature as it helps prevent bad requests due to timezones.

It appears that your station web service is appending `00000Z` to the time string without first checking to see if the `Z` is already present, resulting in an invalid time. Maybe a check to verify first would be more robust.

```
Error 400: Bad Request
schema: error converting value for "startbefore". Details: parsing time "2024-10-31T23:59:59.999Z00000Z": extra text: "00000Z" (and 1 other error)
Usage details are available from https://www.geonet.org.nz/data/tools/FDSN
Request:
/fdsnws/station/1/query?endafter=2010-01-01T00:00:00.000Z&level=station&network=NZ&startbefore=2024-10-31T23:59:59.999Z
Request Submitted:
2024-11-22T11:14:35+13:00
Service version:
1.1
```

junghao, 2025-03-13:
Update deployed.

crotwell, 2025-03-13:
Works for me, thank you.

closed 2025-03-13 by crotwell