GeoNet/help #129: API documentation incorrectly label some endpoints as returning GeoJSON instead of JSON2

carolinaisslaying, 2025-03-06:
## Problematic behaviour
The key issue rests with the API documentation ([api.geonet.org.nz](https://api.geonet.org.nz)) not actually reflecting the correct response type headers for several endpoints.

The following endpoints state that their headers should accept GeoJSON, when the headers actually return json2.
* `/quake/stats`
* `/network/fdsn/station`
* `/network/gnss/mark`

In addition each endpoint's response isn't actually valid GeoJSON, it is in fact json2. Therefore the API response headers are in fact actually correct being json2, but the documentation is incorrect by stating that they are GeoJSON itself.

### Reproducing this issue
Reproducing this issue is pretty simple, you can either make an HTTP request using a package such as `axios` and observe the headers, or even just check the network tab in a web browser and see that the response headers are in fact json2, and not GeoJSON as stated on the documentation.

## Expected behaviour
The API documentation should correctly reflect what each API endpoint returns as best practice by updating the documentation to state that each of the endpoints return json2, and not GeoJSON. This lead to issues when I was creating an API wrapper for GeoNet ([carolinaisslaying/geonet](https://github.com/carolinaisslaying/geonet)).

elidana, 2025-03-06:
Kia ora @carolinaisslaying ,
thanks for this detailed report. We will pass this to our developers and engineers team.

sue-h-gns, 2025-03-25:
We have updated the api documentation endpoint to match the behaviour, as requested.
Thank you again @carolinaisslaying for the report

closed 2025-03-25 by sue-h-gns